frankarobotics/franka_ros #123: Is franka_ros available in Noetic?

YijiongLin, 2021-01-13:
Hi all,

I'd like to confirm that if `franka_ros` in Noetic (the `noetic-devel` branch) has been released with the full functionalities likes the ones in Kinetic and Melodic. Because I prefer to use Noetic but before reinstalling Ubuntu from 18 to 20, I would like to know this question.

If not, which functions could not be used?

All the best,
Bourne

sgabl, 2021-01-13:
Hi,

`franka_ros` is not yet fully released for Noetic. See [here](https://github.com/ros/rosdistro/pull/26417) for more information.

YijiongLin, 2021-01-13:
Hi @sgabl ,

Huge thanks for your kind reply!!! In here it said: 
> Not releasing 'franka_ros' and 'franka_example_controllers' for now because they are depending on MoveIt, which is not packaged yet.
> Cc @davetcoleman, [frankaemika/franka_ros#104](https://github.com/frankaemika/franka_ros/issues/104)

But if I understand things correctly, [Moveit in Noetic](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) has been released. And this [issue 104](https://github.com/frankaemika/franka_ros/issues/104) has been closed. So I'm not sure why `franka_ros` is still not fully released. Could you please help to clarify it?

And if I don't plan to use franka_ros via moveit, is it possible to use the franka_ros directly with full functionalities?

sgabl, 2021-01-13:
Hi,

only the meta-package `franka_ros` and the `franka_example_controllers` which both depend on the (not yet released) `panda_moveit_config` are not yet available.
The others like `franka_hw` and `franka_description` are already available in the ROS repos!

closed 2021-01-13 by YijiongLin

YijiongLin, 2021-01-13:
Hi @sgabl ,

Thanks so much for your time Simon! 

Actually why I thought the `franka_ros` in Noetic is fully released is because in [here](https://index.ros.org/p/franka_ros/github-frankaemika-franka_ros/#noetic) it's just labeled with developed and released (just look the same as the page of Melodic in the ros index website). And in the [Readme file](https://github.com/frankaemika/franka_ros/tree/noetic-devel) there is no clue that franka_ros in Noetic is not fully released yet. It might be kinda misleading.