frankarobotics/franka_ros #128: Is there a way I can view commanded joint torques?

peasant98, 2021-01-26:
Hi, 

Let's say I'm using a Velocity Joint Interface. Is there any way to the joint torques that end up getting sent to the robot!

Thanks!

peasant98, 2021-03-26:
Any update on this @sgabl?

sgabl, 2021-04-07:
Hi, yes you could just check [tau_J_d](https://frankaemika.github.io/libfranka/structfranka_1_1RobotState.html#a7086a89a2705810f93a3a95d43df2d9d) - there you see the desired link-side joint torque.

peasant98, 2021-04-08:
Yes, but if I am running joint velocity control, I would be able to see what's actually being sent.

marcbone, 2022-03-14:
For velocity control there is no direct way to see the commanded torque. To do that you would have to write your own velocity controller on torque basis. However, you can still view the measured torques tau_J. This could be enough for some use-cases.

closed 2022-03-14 by marcbone