frankarobotics/franka_ros #129: xacro:unless with ${not variable}

ndahn, 2021-01-26:
In [this xacro](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_description/robots/panda_arm.xacro) (line 6) `xacro:unless` is used together with `${not connected_to}`. While this is valid syntax, this is a degenerate edge case that will lower compatibility with 3rd party tools (e.g. [xacro-parser](https://github.com/gkjohnson/xacro-parser). This could be easily resolved by using `xacro:if` and `${variable}` instead. I can provide a pull request if desired.

gollth, 2021-09-28:
Hi @ndahn 

sorry for the late reply. I tested to parse the `panda_arm.xacro` with the xacro-parser and got a `XMLDocument` out of it. What exactly doesn't work?

According to the documentation of [xacro](http://wiki.ros.org/xacro#Math_expressions):

> Since ROS Jade, Xacro employs python to evaluate expressions enclosed in dollared-braces, ${}. This allows for more complex arithmetic expressions.

IMO any library claiming to parse xacro, should be able deal with these expressions, including `not`s 😉 So maybe this is rather a  ticket in the [xacro-parser](https://github.com/gkjohnson/xacro-parser) than here?

gollth, 2021-11-09:
I will close this issue here. Feel free to reopen, when there is a bug in our xacro files.

closed 2021-11-09 by gollth