frankarobotics/franka_ros #130: Behavior of different gripper action servers unexpected and inconsistent

fjulian, 2021-03-03:
We're trying to grasp objects with unknown width using the standard Panda gripper. For this, we tried both the `GraspAction` and the `GripperCommandAction` described [here](https://frankaemika.github.io/docs/franka_ros.html#franka-gripper).

The `GraspAction` closes, then keeps applying a force, which is desired. However, the command seems to keep running. When aborting it using the `StopAction` and then trying to open it using  `GraspAction` but the full width, it tries to close again, making it impossible to release the object. From this, we found out the this action server seems to ignore the `width` argument entirely. When sending the same width command multiple times in a row, the gripper opens and closes alternatingly, which is probably undesired.

The `GripperCommandAction` instead honors the width argument that it is passed. However, this one doesn't keep applying the force when reaching the width is impossible due to an object being present between the fingers. This causes the object to slip out of the fingers when lifting the arm. In https://github.com/frankaemika/franka_ros/issues/16 someone explained that this can be avoided by setting the correct width. However, for grasping unknown objects, this is not feasible.

A hack to work around this is to use the `GraspAction` to close the fingers, and the `GripperCommandAction` to open the again. It works for us, is however very unintuitive. Ideally, the `GraspAction` would follow the commanded width and apply the desired force, so that this hack would not be necessary.

Versions of software we use:
Ubuntu 18.04
ROS melodic
libfranka version 0.8.0
franka_ros branch noetic-devel
Franka panda firmware: 4.0.4

gollth, 2021-09-27:
Hello there and thanks for your interest in franka_ros. 

You are already on the right track. The issue is the way the `Grasp` action works. Grasp closes the fingers _as far as possible_ until a contact is detected. Only then it will check if the with between the fingers is close (i.e. within the tolerance `epsilon` you specified) to the `width` you specified. If this condition is met, the gripper keeps applying the force thus holding any object between the fingers. Otherwise it will not exert any forces since it is assumed you grasped something _unintentional_. 

If you want to bring the fingers to a predefined width, please use the `Move` action instead.

With the `epsilon` parameter you can essentially specify the _uncertainty_ about the object you want to grasp. In the case you don't know anything about your object's size you can send an action goal with a width of _4cm_ and an `epsilon` tolerance of _+/-4cm_:

```bash
rostopic pub --once /franka_gripper/grasp/goal franka_gripper/GraspActionGoal "{ goal: { width: 0.04, speed: 0.1, force: 5, epsilon: { inner: 0.04, outer: 0.04 }}}"
```

 This means, that the check above will always succeed for any object between the fingers. Whenever you are done grasping either call `Stop` or `Move` depending on your application.

Does this make things clearer to you?

gollth, 2021-12-27:
I hope this could solve your issue, feel free to reopen if problems persist

closed 2021-12-27 by gollth