frankarobotics/franka_ros #134: Set cartesian impedance failed while running Moveit!

feimeng93, 2021-04-30:
Hi all,
I am implementing impedance control for a trajectory planned by Moveit!. I have tried two approaches but both of them failed:
1. Launch `panda_control_moveit_rviz.launch` from the package `panda_moveit_config` ([https://github.com/ros-planning/panda_moveit_config]) and call service `/franka_control/set_cartesian_impedacne`. However, an error appears as below
```
set_joint_impedance failed: libfranka: Set Cartesian Impedance command rejected: command not possible in the current mode!
```
2. Launch two controllers `position_joint_trajectory_controller` and `cartesian_impedance_example_controller` with the node `controller_spawner` at the same time. As a result, Moveit! can generate a trajectory and it can be executed in rviz. However, it cannot be executed by the real franka robot! During the fake execution period, the topics `/execute_trajectory/actiontopics` and `/position_joint_trajectory_controller/follow_joint_trajectory/actiontopics` both show messages.

Thus, does anyone can help me to tackle these two situations? Any other methods to realize planning with Moveit! and follow the trajectory with impedance control are welcomed! Really appreciate your reading!

Best regards,
Fei

gollth, 2021-12-27:
Hi @feimeng93,

> 1. Launch panda_control_moveit_rviz.launch from the package panda_moveit_config ([https://github.com/ros-planning/panda_moveit_config]) and call service /franka_control/set_cartesian_impedacne. However, an error appears as below

This is expected behaviour. If you look at the [documentation of the FCI](https://frankaemika.github.io/docs/libfranka.html) you will find the following section: 

> Non-realtime commands are blocking, TCP/IP-based and always executed outside of any realtime control loop.

Meaning by design you cannot call the non-realtime commands like set impedance while having an active controller running. To mitigate use something like this:

```bash
rosservice call /controller_manager/switch_controller "stop_controllers: ['<YOUR_CONTROLLER_NAME>']"
rosservice call <YOUR_NON_REALTIME_COMMAND>
rosservice call /controller_manager/switch_controller "start_controllers: ['<YOUR_CONTROLLER_NAME>']"
```

If you are using the [CartesianImpedanceExampleController](https://github.com/frankaemika/franka_ros/blob/develop/franka_example_controllers/include/franka_example_controllers/cartesian_impedance_example_controller.h) you can set the impedances also via dynamic reconfigure without stopping the controller.

> 2. Launch two controllers position_joint_trajectory_controller and cartesian_impedance_example_controller with the node controller_spawner at the same time. 

What would you expect to happen when you run a joint position controller and _simultaneously_ a Cartesian impedance controller? Both controllers claim some command interfaces of the hardware so the controller manager from ROS control shouldn't even allow you to start both. I don't quite understand what you are trying to achieve here, TBH. I suggest you only start one (commanding) controller and use the interface from that. If you need both, first stop the one controller and then start the new one.

For more details on the topic please familiarize yourself again with: http://wiki.ros.org/ros_control

Hope this helps. Feel free to reopen if the issue persists

closed 2021-12-27 by gollth

matthias-mayr, 2023-02-08:
For completeness, we have implemented a [Cartesian impedance controller](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) that can run (MoveIt) trajectories using the nullspace stiffness component while staying compliant.

This way, one does not need to switch to position control & I think this is what the original author wanted to achieve.