frankarobotics/franka_ros #135: Compilation on Windows fails

Tobias-Fischer, 2021-05-17:
Hi there,

We tried to compile the franka_ros stack on Windows, but failed due to this error:
```
FAILED: CMakeFiles/franka_gripper_node.dir/src/franka_gripper_node.cpp.obj 
C:\PROGRA~2\MICROS~1\2019\ENTERP~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DNOMINMAX -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"franka_gripper\" -DWIN32_LEAN_AND_MEAN -D_USE_MATH_DEFINES -Idevel\include -IC:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\include -IC:\Miniconda\conda-bld\ros_1621225433426\_h_env\Library\include -I%PREFIX%\Library\include -IC:\Miniconda\conda-bld\ros_1621225433426\_h_env\Library\share\xmlrpcpp\cmake\..\..\..\include\xmlrpcpp -IC:\Miniconda\conda-bld\ros_1621225433426\_h_env\Library\include\libfranka /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /D _VARIADIC_MAX=10 /Zc:__cplusplus -std:c++14 /showIncludes /FoCMakeFiles\franka_gripper_node.dir\src\franka_gripper_node.cpp.obj /FdCMakeFiles\franka_gripper_node.dir\ /FS -c C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp
C:\Miniconda\conda-bld\ros_1621225433426\_h_env\Library\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp(94): error C3481: 'homing_action_server': lambda capture variable not found
C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp(103): error C3481: 'stop_action_server': lambda capture variable not found
C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp(111): error C3481: 'move_action_server': lambda capture variable not found
C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp(119): error C3481: 'grasp_action_server': lambda capture variable not found
C:\Miniconda\conda-bld\ros_1621225433426\work\ros-noetic-franka-gripper\src\work\src\franka_gripper_node.cpp(127): error C3481: 'gripper_command_action_server': lambda capture variable not found
```

It seems that Visual Studio does not like the lambda expressions.

/cc @wolfv

fcc315, 2022-09-06:
Hi, did you manage to solve this issue?

Tobias-Fischer, 2022-09-06:
No, we haven't continued this any further

fcc315, 2022-09-13:
I'm now free from this error by using "catkin_make -std=c++14".