frankarobotics/franka_ros #136: permission problem on move_to_start?

TracyLestrade, 2021-05-20:
error occurred on running `roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 load_gripper:=false`
enviroments ubuntu 18.04. (realtime kernel).
**libfranka: unable to set realtime scheduling: Operation not permitted**
so I follow this page [](https://frankaemika.github.io/docs/installation_linux.html) and do some operations **Allow a user to set real-time permissions for its processes**
but It didn't work.
`
NODES
  /
    controller_spawner (controller_manager/spawner)
    franka_control (franka_control/franka_control_node)
    joint_state_desired_publisher (joint_state_publisher/joint_state_publisher)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    move_group (moveit_ros_move_group/move_group)
    move_to_start (franka_example_controllers/move_to_start.py)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    state_controller_spawner (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311

process[franka_control-1]: started with pid [20864]
process[state_controller_spawner-2]: started with pid [20865]
process[robot_state_publisher-3]: started with pid [20866]
process[joint_state_publisher-4]: started with pid [20872]
process[joint_state_desired_publisher-5]: started with pid [20873]
process[controller_spawner-6]: started with pid [20879]
process[move_group-7]: started with pid [20882]
process[move_to_start-8]: started with pid [20883]
[ WARN] [1621481675.560030941]: FrankaHW: 
	panda_joint7: 2.808139 degrees to joint limits (limits: [-2.897300, 2.897300] q: -2.848289) 
[ INFO] [1621481675.583559831]: Loading robot model 'panda'...
[ INFO] [1621481675.721385424]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1621481675.728389256]: MoveGroup debug mode is OFF
Starting planning scene monitors...
[ INFO] [1621481675.728420072]: Starting planning scene monitor
[ INFO] [1621481675.729708701]: Listening to '/planning_scene'
[ INFO] [1621481675.729752251]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1621481675.733608172]: Listening to '/collision_object'
[ INFO] [1621481675.734918210]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1621481675.746514535]: Listening to '/camera/depth_registered/points' using message filter with target frame 'world '
[ INFO] [1621481675.749258594]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
[ INFO] [1621481675.779846717]: Initializing OMPL interface using ROS parameters
[ INFO] [1621481675.795302691]: Using planning interface 'OMPL'
[ INFO] [1621481675.797613197]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1621481675.797927168]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1621481675.798171316]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1621481675.798427189]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1621481675.798665960]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1621481675.798907079]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ERROR] [1621481675.799026510]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds
[ INFO] [1621481675.799087234]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1621481675.799118572]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1621481675.799145735]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1621481675.799172189]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1621481675.799198610]: Using planning request adapter 'Add Time Parameterization'
[ WARN] [1621481675.832918380]: Unable to update multi-DOF joint 'virtual_joint': Failure to lookup transform between 'world' and 'panda_link0' with TF exception: "world" passed to lookupTransform argument target_frame does not exist. 
[INFO] [1621481675.986422]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1621481675.988144]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1621481675.989752]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1621481675.991171]: Loading controller: franka_state_controller
[INFO] [1621481675.997402]: Controller Spawner: Loaded controllers: franka_state_controller
[INFO] [1621481676.001319]: Started controllers: franka_state_controller
[INFO] [1621481676.021805]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1621481676.023329]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1621481676.024854]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1621481676.026443]: Loading controller: position_joint_trajectory_controller
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[INFO] [1621481676.054227]: Controller Spawner: Loaded controllers: position_joint_trajectory_controller
[ INFO] [1621481676.056451292]: FrankaHW: Prepared switching controllers to joint_position with parameters limit_rate=1, cutoff_frequency=100, internal_controller=joint_impedance
terminate called after throwing an instance of 'franka::RealtimeException'
  what():  libfranka: unable to set realtime scheduling: Operation not permitted
[INFO] [1621481676.057228]: Started controllers: position_joint_trajectory_controller
================================================================================REQUIRED process [franka_control-1] has died!
process has died [pid 20864, exit code -6, cmd /opt/ros/melodic/lib/franka_control/franka_control_node __name:=franka_control __log:=/home/mv1804/.ros/log/ea63eb72-b91b-11eb-b61c-54bf64915687/franka_control-1.log].
log file: /home/mv1804/.ros/log/ea63eb72-b91b-11eb-b61c-54bf64915687/franka_control-1*.log
Initiating shutdown!
================================================================================
`

gollth, 2021-12-27:
I assume you could solve your problem, feel free to reopen this issue persists

closed 2021-12-27 by gollth

YummyOu, 2022-03-07:
Hi, I also meet this problem.
How could you solve it?

Test / also use somehow roslaunch commands
`./communication_test 172.16.0.2`
Show
```
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...
 
libfranka: unable to set realtime scheduling: Operation not permitted
```

marcbone, 2022-03-07:
@YummyOu Your problem is that you do not have a real-time kernel or it is not properly set up. Here you can find instructions on how to [set up a real-time kernel](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel)

YummyOu, 2022-03-07:
All of these commands have already been checked:
- uname -a
- groups
- cat /etc/security/limits.conf

[image]

gavanderhoorn, 2022-03-07:
Did you also reboot / log-out -> log-in?

Your user doesn't appear to be a member of the `realtime` group. This is a requirement, if you followed [Allow a user to set real-time permissions for its processes](https://frankaemika.github.io/docs/installation_linux.html#allow-a-user-to-set-real-time-permissions-for-its-processes).

YummyOu, 2022-03-07:
Hi @gavanderhoorn! My username is "elfin", is that the first one of `groups` of the above picture?

YummyOu, 2022-03-08:
I solved the problem!

@gavanderhoorn Thx! You are right. 

I didn't add my user in the `realtime` group. When I check `getent group`, it just showed `realtime:x:1002`. So I tried `sudo usermod -a -G realtime elfin` and rebooted. Finally, works:)

Then, I met a problem called `communication_constraints_violation`. I just changed a cable of the network and it works.

The robot moved to the home position and showed some `current success rate: 0.9x` interminal.

maxiuw, 2022-09-17:
very simple check (i totally missed that), eventhough my pc usually boots in rt kernel, it didnt so i spent 2h trying to fix, where later i realized it was booted with wrong kernel. Choose `ubuntu advanced` when booting ant make sure that the kernel is an `rt` kernel

Amar-Judo, 2022-11-29:
> 

Hello, 
I am Amar, I am new to this platform. 

this is my error.

**user@user-ESPRIMO-P420:~/libfranka/build/examples$ communication_test 172.16.0.2
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

libfranka: unable to set realtime scheduling: Operation not permitted**

I am attaching my screenshots of the commends 

-     uname -a
-     groups
-     cat /etc/security/limits.conf

[image]
[image]

gollth, 2022-11-29:
Hi @Amar-Judo,

please when asking questions always post your outputs in _text_ form, e.g. by copy pasting them in code blocks. 

Regarding your problem: From your `/etc/security/limits.conf` file it looks like you added the realtime constraints, but commented them out. Try removing the leading `#` in front of them:

```python
# Notice, no '#' at the beginning of each line
@realtime soft rtprio 99
@realtime soft priority 99
@realtime soft memlock 102400
@realtime hard rtprio 99
@realtime hard priority 99
@realtime hard memlock 102400
```

For more information follow closely https://frankaemika.github.io/docs/installation_linux.html#allow-a-user-to-set-real-time-permissions-for-its-processes

Amar-Judo, 2022-11-29:
> Hi @Amar-Judo,
>
> please when asking questions always post your outputs in _text_ form, e.g. by copy pasting them in code blocks.
>
> Regarding your problem: From your `/etc/security/limits.conf` file it [...]

yes, it is working now. 

Thanks