frankarobotics/franka_ros #143: About the values of physics related parameters in panda_gazebo.xacro

vpadois, 2021-08-06:
Thanks for (finally) providing an official URDF file for your robot. Yet, I find it quite amazing that this file does not contain Franka Emika's estimate of the physics related parameters (e.g. inertia tensors for each body) but instead rely on the values coming from an external paper (which everybody was using in their hand made URDF until you released yours). 

While the values coming from this paper provide an overall good approximations of the dynamics of the robot, the state-of-the-art identification approach on which it relies tends to lead to a distribution of the identification errors which is not in favour of the smaller joints (e.g. their associated inertia tensor ends-up underestimated). Unfortunately, these are the joints where the ratio between inertial torques (the ones you need to accelerate each body) and friction torques (among which dry friction) is the smallest. 

If your inner torque loop was perfectly rejecting friction related disturbances, that should not be too much of a problem. Unfortunately it does not and while everything is perfect in simulation where dry friction is not simulated and where the simulation model is the same as the control one, the fact of underestimating these inertia values (especially Izz) is really problematic with the real robot. This can lead to bad control performance when you, for example, want to implement control laws implying the use of the mass matrix (e.g. such as computed torque control). 

Of course, one could be using the inertia matrix provided by libFranka but :
- this does not actually fully solve the problem as we cannot really know how well the small joint inertia values are identified there.
- it means developing robot specific controllers which is far from ideal from a genericity point of view.
- it means testing your controller in simulation and on the real robot with to different models. Clearly this is not the best thing to do.

So, I would have two requests related to this issue:
- please provide a model including the physics related parameters values YOU are using internally. I do not see any good reason not to do it. If you do not do so, you should not claim that your provide the "access to kinematic and dynamic robot model " here  [https://www.franka.de/academia/](https://www.franka.de/academia/). This is false advertising.
- please provide the characteristics or some straightforward way to monitor the inner control torque loop performances. How can on develop torque control laws when the inner torque loop performance is not known?

sabaghian, 2021-08-24:
Thank you for you comment. As we are addressing a wide spectrum of researchers, we need to provide interfaces which are the most convenient and bring the most use to the community. Franka does provide access to kinematics and dynamics model of the robot using M,c,g matrices which are the most convenient interfaces to use (according to our research).
While dynamic models from estimated CAD model of the robot do not usually provide the desired accuracy, the identified models usually deliver sufficient accuracy. However, the identified model is usually a function of the input signals. Since the signals are filtered and processed differently in different control levels, it is more suitable for FCI users to use the model identified using FCI signals. Our ultimate goal is to establish a research community and contribute to the robotics community by building upon all researchers' work. Therefore, we found this identified model the most accurate one for the FCI users. We are of course, open to replace it with more accurate ones identified by you or other talented researchers in the community.

vpadois, 2021-09-06:
Thanks for your answer. In all honesty, I believe that the argument regarding the fact that :
- the API is designed to bring convenience to the research community;
- the proposed identified models usually deliver sufficient accuracy;
are fallacious.

Research is about better understanding and going beyond existing limits. How can that be achieved with robots which for most part remain technological black boxes with no or very approximate capabilities information? Even though you provide more information and deeper access to low-level control than most robots' manufacturers, what you provide is still very incomplete. Also,  what is sufficient accuracy?  Accuracy of what? Torque/Force control, position control, velocity control? In Cartesian space? In joint space?

Finally, you suggest that we should perform better identification of the dynamics of the system. Doing so would actually require to include actuation and transmission in the model but, even more importantly, it would also require you to provide us with access to motor side physical quantities. Obviously you are not willing to do so.

Best
Vincent

gollth, 2021-09-14:
Hi @vpadois,

thank you for your contribution and interests in providing a narrower gap between simulation and real hardware. We really appreciate that. However, I fear that you are asking is out of the scope of `franka_ros` and `libfranka`. This is more a political discussion about what can be made open source and what not. 

I will close this issue here since there is nothing we can do in `franka_ros` to help you. Please feel free to bring up your points in https://www.franka-community.de/ under the FCI topic, though!

All the best

closed 2021-09-14 by gollth

vpadois, 2021-09-14:
Hi @gollth,

Thanks for your suggestion. I did bring up this topic on the 19th of September 2018 in the Franka community forum (https://www.franka-community.de/t/urdf-model-of-the-panda-robot-including-masses-for-simulaton-in-gazebo/200).

I never received any specific answer from people at Franka Emika (neither did I from most robot manufacturers such as Kuka when I was using a LWR IV+).

I understand that this topic is highly political. Nevertheless, as much as I really truly appreciate the leap forward Franka Emika has allowed researchers to make, I am still deeply convinced that most advances made in Robotics over the last 50 years were vastly possible through the investment of public money. Most companies benefit largely from this money directly or indirectly as well as from the research results generated from public labs. Yet, this is mostly a one way process. This is highly frustrating and I will keep equally praising companies such as Franka Emika for their great products and complain about the impossible access to highly necessary information by researchers. So be ready to read my complaints from time to time ;)

Best
Vincent