frankarobotics/franka_ros #160: Gazebo model joints move by themselves

saalimperator, 2021-08-29:
Hello!

I'm currently working with your Gazebo model to test my controller. To do this, I modified one of your example controllers so that all joints except the first one keep their current angle. The first one is waiting for input from my controller, which is applied directly to the joint using the ".setCommand(tau)" function. 

Now for my problem: when I start Gazebo, it seems to maintain the initial state, but after a few seconds the first joint starts to move slightly counterclockwise without any input from me. Is this normal behavior of this joint when the input is set to zero? Or is there something wrong with the inertia, friction and damping of this joint. When I send some inputs to this joint, it starts to move, but it is not slowed down by friction or damping and just bounces back when it hits the joint limits.

Can you give me a hint where I am doing something wrong?

rickstaa, 2021-09-17:
@saalimperator Thanks for creating this issue. I also noticed this behaviour even when no controllers are loaded. In that case, one would expect the robot to just fall down and stay there. The behaviour that occurs is however very strange (See gif 1 below). 

[image]

It could be related to the unstable dynamics that are described in https://github.com/justagist/franka_panda_description/issues/5.

gollth, 2021-09-30:
> In that case, one would expect the robot to just fall down and stay there.

Not really. The `franka_hw_sim` plugin simulates the real robot in that it compensates the gravity of arm when you send zero (or no) commands from the controller.

https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gazebo/src/franka_hw_sim.cpp#L310-L317

Imagine this on the real robot. When you have a ROS controller which sends zero torque, the robot won't fall to the ground either but behaves like in "free space" essentially. This is also what happens when no controller is running.

Why the robot is drifting is unclear to me, though. I suspect numerical instabilities somewhere between `franka_hw_sim` and Gazebo.

> when I start Gazebo, it seems to maintain the initial state, but after a few seconds the first joint starts to move slightly counterclockwise without any input from me.

@saalimperator  a possible solution would be to have the controller active at all times. So while you are waiting for input, generate commands in your controller that hold the robot at a defined configuration. This should stop the drifting

rickstaa, 2021-09-30:
> Not really. The `franka_hw_sim` plugin simulates the real robot in that it compensates the gravity of arm when you send zero (or no) commands from the controller. [...]

@gollth At the time of writing, I forgot to check the [franka_hw_sim](franka_ros/franka_gazebo/src/franka_hw_sim.cpp) for any gravity compensation. Thanks for pointing me to the relevant code. In that case, numerical instabilities could indeed cause this drifting. I have seen the same behaviour in other simulated robots.

For now, I will use your workaround and try to send control commands to mitigate the drift.

saalimperator, 2021-09-30:
> @saalimperator a possible solution would be to have the controller active at all times. So while you are waiting for input, generate commands in your controller that hold the robot at a defined configuration. This should stop the drifting

Thanks for the answer. I already expected something like this. Unfortunately, your proposed solution does not work for my project, because I am trying to determine the complete dynamics of the robot by means of an experiment. I have already found a better simulation environment for my case. Much success with the programming!

rickstaa, 2021-09-30:
@saalimperator May I ask which simulation environment you switched? 

Maybe, it is an idea to keep this issue open as a bug report? In that case, @emikafranka can keep it open or close it, depending on whether it is placed on their backlog.

saalimperator, 2021-10-01:
Sorry I got that wrong, I didn't switch to another environment I switched the model. I am now working with an UR in Gazebo. But what I want to do is rather special. My programm communication looks like this:
Matlab<->Simulink<->ROS<->Gazebo

rickstaa, 2021-10-01:
@saalimperator Thanks for your response! Ah, I see colleagues of my have the same communication scheme. 

Since the issue you described above still exists in the `franka_gazebo` package, maybe let's re-open it for now? In this way, it can serve as a bug report.

gollth, 2021-11-05:
Okay I think I have an idea where the problem comes from. It's part of a systematic error in the current calculation of the _efforts_ coming from Gazebo. It's **not only** related to joints moving on their own without controller, but **will affect any controller**!

Let me try to explain what I think is going on:

## Hypothesis

**There is a bug in the way Gazebo calculates torques if the joint origin is rotated**. The critical function is [`gazebo::physics::ODEJoint::GetForceTorque()`](https://github.com/osrf/gazebo/blob/gazebo9/gazebo/physics/ode/ODEJoint.cc#L651-L849)

## Bug Reproduction

### Nominal Case

To see more clearly what is going on, let's create a simplified URDF containing only equally long sticks (length _10cm_) with equal masses (_5kg_), where joints have zero rotation (i.e. `<origin rpy="0 0 0" />`) and bring the robot in the following configuration: 

[image]

Let's focus on `joint6` individually. We would expect a (negative) torque around `joint6`'s Y axis because of gravity of link 6 and 7. If we print some values from during the `writeSim()` step, we see this confirmed:

| Quantity                    | Value                               | Source                                                                                                                                                          |
| :--                         | :--                                 | :--                                                                                                                                                             |
| Effort acting on joint      | `-0.750765`                         | `body2Torque.dot(joint->axis)`                                                                                                                                  |
| KDL Gravity Term            | `0.750765`                          | [`model->gravity(q)`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L310)                                          |
| Torque calculated by Gazebo | `[-0.000793, -0.750765, -0.225921]` | [`GetForceTorque(0).body2Torque`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/joint.cpp#L32)                                       |
| Joint Axis in World frame   | `[0.000000, 1.000000, 0.000000]`    | [`joint->GetGlobalAxis()`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/1.9.1/classgazebo_1_1physics_1_1Joint.html#a9bf1a657e0b9e56999f080d328ae3557) |
| Joint Axis in Local frame   | `[0.000000, 1.000000, 0.000000]`    | [`joint->GetLocalAxis()`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/1.9.1/classgazebo_1_1physics_1_1Joint.html#a9b665ad21bf47509833987ddeed99e40)  |
| Joint Axis defined in URDF  | `[0, 1, 0]`                         | [`joint->axis_`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L79)                                                |

As expected Gazebo measures a torque around the Y-axis of the joint, which is the degree of freedom, and some stray torques in other directions. The magnitue also seems to match. We have the torque from link6 (_5kg × 0.05m ≈ 0.25Nm_) plus the torque from link7 (_5kg × 0.1m ≈ 0.5Nm_). This matches the torque which KDL calculates for its gravity term.

### Bugged Case

Now let's do the same thing, but *rotate* the joint's origin by 90 degrees, i.e. `<origin rpy="${pi/2} 0 0" />`. Bringing the robot in a similar configuration should now make it experience (negative) torque, this time only coming from the weight of link7 alone, since `joint6` is now rotating around Z.

[image]

Again we look at the interessting values:

| Quantity                    | Value                              | Source                                                                                                                                                          |
| :--                         | :--                                | :--                                                                                                                                                             |
| Effort acting on joint      | `-0.507839`                        | `body2Torque.dot(joint->axis)`                                                                                                                                  |
| KDL Gravity Term            | `0.250255`                         | [`model->gravity(q)`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L310)                                          |
| Torque calculated by Gazebo | `[-0.000254, -0.507839, 0.000003]` | [`GetForceTorque(0).body2Torque`](https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/joint.cpp#L32)                                       |
| Joint Axis in World frame   | `[0.000000, -1.000000, 0.000000]`  | [`joint->GetGlobalAxis()`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/1.9.1/classgazebo_1_1physics_1_1Joint.html#a9bf1a657e0b9e56999f080d328ae3557) |
| Joint Axis in Local frame   | `[0.000000, -1.000000, 0.000000]`  | [`joint->GetLocalAxis()`](https://osrf-distributions.s3.amazonaws.com/gazebo/api/1.9.1/classgazebo_1_1physics_1_1Joint.html#a9b665ad21bf47509833987ddeed99e40)  |
[... cut ...]
For the original Panda URDF this get's of course more complicated. When we plot the difference between the gravity term of KDL and the torque measured by Gazebo over time we see that this is really a systematic error and not (only) related to numerical noise as initially assumed:

[image]

## Solutions

1. Create a bug report in [osrf/gazebo](https://github.com/osrf/gazebo/issues), however there are 1.4k other issues already waiting...
2. Transform meshes/frames/inertias/axis such that the joint in the URDF do not contain any rotation.
3. Hope anybody has an idea how to correctly transform the result of `GetForceTorque()`. Would appreciate any further ideas or insights 😕

gavanderhoorn, 2021-11-05:
>     1. Create a bug report in [osrf/gazebo](https://github.com/osrf/gazebo/issues), however there are 1.4k other issues already waiting...

I would suggest to *always* do this, if you believe there is no other bug report already open about this.

The `osrf/gazebo` issue tracker is sort-of also used as a (public) todo-list, so don't let that high nr of open issues dissuade you from reporting something you believe is wrong.

Getting this fixed -- provided the analysis is correct -- would be tremendously valuable.

rickstaa, 2021-11-05:
@gollth Thanks a lot for finding the root cause of this issue and for creating the steps to reproduce it. Can you maybe push a small example repo to GitHub with the files you used to debug this problem? This will save me time when repeating the steps you describe above.

I agree with @gavanderhoorn this problem is likely affecting multiple repositories and users and should be fixed on `osrf/gazebo`. I think @gavanderhoorn's suggestion to check if the problem exists in [ignition](https://github.com/ignitionrobotics) would be an excellent first step. If it does, we can create an issue there and backport the solution. If it does not, we can check what ignition does differently.

I will try to reproduce your example and then check if I can find more information about what causes the issue inside the gazebo/ignition code.

gollth, 2021-11-08:
Hey @rickstaa, thanks for taking a look. You can find the Stick URDF and the prints in [this commit](https://github.com/frankaemika/franka_ros/commit/786f03a4db31f138aa19a7d25b0fe0befed1ef4f). Also look at its commit message to see how to reproduce the above described behavior.

rickstaa, 2021-11-26:
@gollth I will start my investigation of this issue. I will keep this comment as a notepad.

### Possibly related issues

- https://github.com/osrf/gazebo/issues/2476
- https://github.com/osrf/gazebo/issues/2209

rickstaa, 2021-11-26:
@gollth quick question. I think

```bash
$(calc 'deg2rad(90)')
```

is syntax that is specific to your setup? I think `calc` is referring to the [calc-common](https://packages.ubuntu.com/search?keywords=calc-common) package but I could not find the `deg2rad` function. I will use hardcoded values for now.

rickstaa, 2021-11-26:
@gollth Based on your commit, I created a minimal example repository (i.e. no dependency on franka_ros) we can use in our Gazebo issue. This repository can be found [here](https://github.com/rickstaa/gazebo-torque-calculation-bug). I will make sure an RQT plot displays the torque, the joints can be controlled through a GUI, and some information is printed to the cmd line. I will keep you posted.

gollth, 2021-11-29:
> @gollth quick question. I think
> 
> ```shell
> $(calc 'deg2rad(90)')
> ```
> 
> is a syntax that is specific to your setup? I think `calc` is referring to the [calc-common](https://packages.ubuntu.com/search?keywords=calc-common) package but I could not find the `deg2rad` function. I used hardcoded values for now but I think these methods are very useful when having to deal with non `0/30/45/60/90` degree angles.

Hi @rickstaa,

thanks for looking into the issue. No `calc` is way dirtier than that, and comes from my local setup:

```bash
calc () {
  expression=$@ 
  python -c "from numpy import *; print($expression)"
}
```

Feel free to use whatever else you find suitable

rickstaa, 2021-11-29:
Ha nice little trick! I mostly write bash scripts but this is a lot faster and versatile.

rickstaa, 2021-11-30:
@gollth I investigated your issue, but I'm not able to reproduce it using your steps provided in https://github.com/frankaemika/franka_ros/issues/160#issuecomment-961780423. Maybe I misunderstood your example, or I overlooked something.

What I noticed at first is that it looks like the figure included in the *Bugged Case* does not correspond with the data shown in the table. In the figure, the local URDF axis is `0 0 1` where the table shows `0 1 0`. Further, I cannot reproduce the error using your example and can not find any problems with Gazebo in my new example (see details below).

#### Your example

You can use my repository since the `franka_ros` submodule that is in there includes your commit. The only thing I did was to add a `rotated` launch file argument to make it easier to inspect both cases (see [c8b153b14dfd27ff513289bdf2f916e343ce66c3](https://github.com/rickstaa/franka_ros/commit/c8b153b14dfd27ff513289bdf2f916e343ce66c3). But the error also doesn't show up on my system when I revert this commit.
I added a `rotated` xacro argument to make it easier to switch in your example. Apart from that the example is the same. I use the following command for the correct case:

```bash
roslaunch franka_gazebo panda.launch \
  use_gripper:=false \
  headless:=false \
  paused:=true \
  initial_joint_positions:="-J panda_joint6 $(calc 'deg2rad(-90)')"
```

and the following for the bugged case:

```bash
roslaunch franka_gazebo panda.launch \
  use_gripper:=false \
  headless:=false \
  paused:=true \
  rotated:=true \
  initial_joint_positions:="-J panda_joint6 $(calc 'deg2rad(90)')"
```

#### My example

You can find a guide on using my example in the [README](https://github.com/rickstaa/gazebo-torque-calculation-bug#readme). To quickly start, run the following command after building the catkin workspace:

```bash
 roslaunch gazebo_torque_calculation_bug stick.launch
```

As you can see, none of the rotations causes a problem.

rickstaa, 2021-11-30:
@gollth The only two things that I have found so far from Gazebos side that could perturb your gravity compensation are:

- There is a startup delay of `0.2` s for the Gazebo effort calculation. This delay is present in both the `GetForce` and `GetForceTorque(0).body2Torque` methods but not for the gravity component that is calculated using [the orocos KDL library](https://www.orocos.org/kdl.html). As the delay is small and subsides quickly, it should not influence the gravity compensation controller much.

[image]

- There also is a minor delay; see the peak in the gif below when we control the robot. As the difference subsides quickly, this shouldn't influence the drift that is seen when the robot is stable.

[image]

rickstaa, 2021-11-30:
I also tried to migrate the example to the Ignition simulator see [gazebo-torque-calculation-bug#ignition-gazebo](https://github.com/rickstaa/gazebo-torque-calculation-bug/tree/ignition-gazebo). I, however, think that this requires a bit more work since the [https://github.com/ros-simulation/gazebo_ros_pkgs](ros-simulation/gazebo_ros_pkgs package) package is not ignition compatible. I could not find an Ignition equivalent in the [ros_ign](https://github.com/ignitionrobotics/ros_ign#packages).

I think that testing the bug in the ignition simulator requires us to create a small sensor plugin that reads the required data from the ignition simulator while finding a way to get the robot in the correct configuration. I think this can be done by extending the [ros_ign_gazebo/create.cpp](https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_gazebo/src/create.cpp) method to take in joint angles similar to what is done in the [gazebo_ros/spawn_model](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_ros/scripts/spawn_model#L78-L79) method.

As I have no experience with the ignition simulator, I am not sure the easiest way to test this bug. Maybe @gavanderhoorn has some suggestions. However, let's first find the proper steps to reproduce the bug in Gazebo.

gavanderhoorn, 2021-12-01:
I'm not sure I completely understand what you're looking for, but perhaps ignitionrobotics/ign_ros2_control#1 helps?

rickstaa, 2021-12-01:
@gavanderhoorn Thanks for your answer. I did not see that pull request; it will undoubtedly be helpful in the future! 

For now, I ended up coding the configurations directly in the world file while using a `rotated` launch argument to switch between the nominal and bugged position. I then use the `ignition-gazebo-forcetorque-system` plugin to get the torque. I then use the `ignition::gazebo::systems::JointPositionController` plugin to control the joints through the `Joint position control` ignition GUI.

rickstaa, 2021-12-01:
@gollth The ignition example can now also be found under [the example repository](https://github.com/rickstaa/gazebo-torque-calculation-bug). Use the following command to start the nominal case:

```bash
roslaunch gazebo_torque_calculation_bug stick_ignition.launch
```

To start the bugged case use:

```bash
roslaunch gazebo_torque_calculation_bug stick_ignition.launch rotated:=true
```

After you started the simulation you can check the joint force and torque by using the following command:

```bash
ign topic -e -n1 -t /stick_joint1/force_torque
```

In both options, you can control the stick joints using the `Joint position controller` GUI.

### Results

#### Nominal case

[image]

```bash
![image](https://user-images.githubusercontent.com/17570430/144325720-0997da5a-fbca-4afb-95f7-47156fee9990.png)
```

#### Bugged case

[image]

```bash
force {
  x: -2.7957245970631345e-05
  y: 5.97209721053419e-19
  z: 9.999998399617791
}
torque {
  x: -0.24999995998758542
  y: 0.74999968617028756
  z: -6.989311451353745e-07
}
```

rickstaa, 2021-12-02:
@gollth I did one last inspection, and it looks like the torque and force calculations are working fine in both the Gazebo 11 and Ignition v6.0 simulators (see gif below). I will add a gravity compensator to the stick and check other things that might go wrong. We can reopen the Gazebo investigation if you find the configurations that caused the problems.

#### Nominal version

[image]

#### Bugged version

[image]

rickstaa, 2021-12-02:
# Gravity compensation investigation

Here I will add the results of my gravity compensation investigation.

## Reproduce gravity compensation bug in my example

## Reproduce in Gollths' example

To test this out in my example, we can use the following commands:

```bash
roslaunch gazebo_torque_calculation_bug stick.launch load_controllers:=false rotated:=true paused:=false
```

or 

```bash
roslaunch gazebo_torque_calculation_bug stick.launch load_controllers:=false rotated:=true paused:=false
```

We can enable/disable the control and control the arm to different positions to make the drift appear.

rickstaa, 2021-12-03:
# Gravity compensation bug investigation

This comment will show the results of my investigation into the bug that is found in the gravity compensation algorithm in your example:

## Bug 1: Wrong gravity vector

The first obvious error has already been discussed above and was fixed when #177 was merged. In the old code the hardcoded gravity vector was different from the actual gravity vector used by Gazebo. After applying #177 onto the example repository see [check_gazebo_gravity_problem-fixed](https://github.com/rickstaa/franka_ros/tree/check_gazebo_gravity_problem-fixed) we got the following behaviour:

## Bug 2:

rickstaa, 2021-12-03:
# Gravity compensation bug investigation

This comment will show the results of my investigation into the bug that is found in the gravity compensation algorithm in your example:

## Bug 1: Wrong gravity vector

The first obvious error has already been discussed above and was fixed when #177 was merged. In the old code, the hardcoded gravity vector was different from the actual gravity vector used by Gazebo. After applying #177 onto the example repository see [check_gazebo_gravity_problem-fixed](https://github.com/rickstaa/franka_ros/tree/check_gazebo_gravity_problem-fixed) we got the following behaviour:

[image]

[image]

[image]

## Conclusion

@gollth, @marcbone I concluded my investigation into the gravity compensator see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982682096, https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982933376, https://github.com/frankaemika/franka_ros/issues/160#issuecomment-984520245 and https://github.com/frankaemika/franka_ros/issues/160#issuecomment-984604733. 

In the end, I could not find any problems with the gravity controller other than the one that I already fixed in #177. I am therefore unsure why the panda robot is drifting when no controllers are enabled but it looks like the gravity compensator is working fine. I have the following hypothesis about why the robot might drift when no controllers are active:

- [ ] Wrong inertias, masses and joint frictions cause the model to be unstable. 
- [ ] Some joints in the robot have a non-zero velocity when the simulation is started. I saw this before in other robots where the gravity controller was started 2-time steps after the robot was launched into a  running simulation. Here, since the gravity compensator is found in the `FrankaHWSim::writeSim`, this should not be the case.

https://github.com/frankaemika/franka_ros/blob/17ea46264f8f982d00d859abe5058d564f9ae3ce/franka_gazebo/src/franka_hw_sim.cpp#L325

rickstaa, 2021-12-03:
# Investigate panda start velocities

## Results

Let's start a great plot with the panda_gazebo simulation to see if there are any starting velocities when the simulation is started. To test this out, let's add an RQT plotter to the `panda.launch` file (see https://github.com/rickstaa/franka_ros/tree/check_gazebo_start_velocity) and start the simulation without the hand:

```bash
roslaunch franka_gazebo panda.launch use_gripper:=false paused:=true
```
Then start the following plots:

```bash
rosrun rqt_plot rqt_plot /joint_states/velocity[0] /joint_states/velocity[1] /joint_states/velocity[2] /joint_states/velocity[3] /joint_states/velocity[4] /joint_states/velocity[5] /joint_states/velocity[6]
```

```bash
rosrun rqt_plot rqt_plot /panda_joint1/error /panda_joint2/error /panda_joint3/error /panda_joint4/error /panda_joint5/error /
panda_joint6/error /panda_joint7/error
```

```bash
rosrun rqt_plot rqt_plot /velocity_norm
```

```bash
rosrun rqt_plot rqt_plot /total_energy
```

[image]

[image]

[image]

[image]

The energy calculations are not theoretically correct serve as quick estimation to check if significant energy is added to the system. When we zoom in on the velocity plot, we can see that the starting velocities are non-zero. They are responsible for the robot movement since the gravity compensator only compensates for gravity. The interaction between the different joints causes the complexity of the movement.

[image]

### Remove start velocities using joint trajectory controller

Let's start a joint_trajectory controller, let it run for 5 seconds, and disable it to see if the drift still exists:

```bash
roslaunch franka_gazebo panda.launch use_gripper:=false paused:=true controller:=effort_joint_trajectory_controller
```

```bash
rosrun rqt_controller_manager rosrun rqt_controller_manager
```

[image]

[image]

[image]

[image]

Stopping the controller gives strange effects; the robot has abrupt movement, but then the robot stays stable.

rickstaa, 2021-12-03:
# Migrate stick to panda

Let's slowly transfer the stick into the panda robot. I will do this on https://github.com/rickstaa/franka_ros/tree/migrate_stick_2_panda. Each commit will represent a migration step:

## Same configuration

## Same joint lengths

## Same axis positions

## Same mass

## Same moment of inertia

## Same joint properties

gollth, 2021-12-07:
Hello @rickstaa,

first of all thanks a lot for your great efforts you put into investigation of this bug. I didn't have yet the time to retrace your whole steps but definitely will in the near future.

Quick comment I noticed right away: I think there is a difference between *Gazebo 9* & *Gazebo 11*. I tried the example from your fork on my machine (Ubuntu 18.04) and got _different_ results than you (I guess you run 20.04?):

```
▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L335 ⟫ panda_joint6: Error:                 0.250258
▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L336 ⟫ panda_joint6: Total Effort:          0.000003
▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L337 ⟫ panda_joint6: KDL GravComp:          0.250255
▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L338 ⟫ panda_joint6: KDL GravComp flipped:  -0.250255
``` 

Notice the error is not zero. I was trying on ROS Noetic (20.04) and there I get similar values [that you posted above](https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982682096). 

> I cannot reproduce the error between the `KDL GravComp` and `Total Effort` using your example

That might also explain why you couldn't reproduce my bugged case. Could you verify that?

rickstaa, 2021-12-07:
@gollth Thanks for working with me on this bug! I suspected a version difference to be the cause of the
reproducibility problem. I, however, did not have time to verify this since my experiments are in Gazebo 11. That is handy information since we can probably back-port the fix from Gazebo 11 to Gazebo 9. I will verify this today so that we can open a Gazebo issue. 

In the meantime, I found the leading case of the resulting movement observed in Gazebo 11. It looks like two things cause it.

- There is a small initial velocity when the simulation is started, causing the robot to drift.
- Further, it looks like the collision meshes do not work nicely together and amplify this movement. When I add them to the robot, the drift is amplified.

I, however, am still performing some experiments before I can conclude. I will update https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985620934 and let you know.

rickstaa, 2021-12-07:
@gollth Thanks for working with me on this bug! I suspected a version difference to be the cause of the
reproducibility problem. I, however, did not have time to verify this since my experiments are in Gazebo 11. That is handy information since we can probably back-port the fix from Gazebo 11 to Gazebo 9. I will verify this today so that we can open a Gazebo issue. 

In the meantime, I found the leading case of the resulting movement observed in Gazebo 11. It looks like two things cause it.

- There are small initial velocities when the simulation is started, causing the robot to drift. The effect of this can be mitigated by increasing the joint damping. I, however, think this is only wise if the joint damping stays close to the actual joint damping.
- Further, it looks like the collision meshes do not work nicely together and amplify this movement. When I add them to the robot, the drift is intensified.

I, however, am still performing some experiments before I can conclude. I will update https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985620934 and let you know.

rickstaa, 2021-12-08:
# Investigate error blowup

Let's quickly investigate what goes wrong in the aforementioned error blowup when the moments of inertias of the panda robot are added to the stick robot.

## Time step and configuration when the blowup happens

rickstaa, 2021-12-08:
# Investigate gravity compensation at joint limits

Let's investigate whether the gravity compensation at the joint limits causes problems. To do this, let's launch all joints close to their joint limits:

```bash
roslaunch gazebo_torque_calculation_bug stick_long.launch paused:=true initial_joint_positions:="-J stick_joint1  -2.967 -J stick_joint2 1.8325 -J stick_joint3 -2.967 -J stick_joint4 -3.1415 -J stick_joint5 -2.967 -J stick_joint6 -0.0872 -J stick_joint7 -2.967"
```

Let's also again launch the position/joint_limit plots and the error plots:

```bash
roslaunch gazebo_torque_calculation_bug plot_joint_positions_and_limits.launch
```

```bash
rosrun rqt_plot rqt_plot /stick/gazebo_bug/stick_joint1/error /stick/gazebo_bug/stick_joint2/error /stick/gazebo_bug/stick_joint3/error /stick/gazebo_bug/stick_joint4/error /stick/gazebo_bug/stick_joint5/error /stick/gazebo_bug/stick_joint6/error /stick/gazebo_bug/stick_joint7/error
```

### Results

This does not work since we have set `safety_controller` soft limits.  We can disable the soft limits using the `safety_controller:=false"

rickstaa, 2021-12-08:
# Investigate effect of safety_controllers

Let's check if the gravity compensation error explosion is removed when we disable the soft joint limits that are applied by the `safety controller`. This can be done by using the `safety_controller:=false` launch file argument.

## Plots

## Conclusion

rickstaa, 2021-12-08:
@gollth, @marcbone I concluded my investigation into the motion drift; you can check the comments above. I added the plots and figures under a collapsable `<details>` section to remove some noise. I summarized my results below:

## Conclusions

-There appears to be a small startup and command delay for the effort coming from Gazebo (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982933376). This delay only seems to be present when no effort control commands are sent. It becomes neglectable when the gravity compensation is enabled and should not be a problem.
-  Gazebo appears to have a slight joint drift when your 7DOF stick is loaded. This drift is likely due to numerical instabilities but is very small and should therefore not influence simulation experiments in which the robot is controlled (see the [the same configuration section of the migration tests](https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985620934)). 
- Both the original long stick and the fully migrated stick have non-zero joint velocities when the Gazebo simulation is started. Altough these velocities are small, I don't think they need to be there. I have not yet found what causes these initial velocities (https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985593607).
- One bug in the current gravity compensation is that the parameters of the `safety_controller` are probably tuned incorrectly (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-988621107).
- Another error in the gravity compensation is that it does not account for the hard joint limits (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-988621107).

## Next steps

The last two points can be solved by improving the `safety_controller` parameters and/or adding a simple check before the gravity compensation is applied. I will leave this to you since there are many ways to implement this. I think solving this will significantly improve the gravity compensation. 

For the rest of the drift, I noticed that it becomes problematic when the collision geometries are added, but I could not yet find out why this is the case. Since I have been working on this issue for a long time, I think a fresh pair of eyes would spot the things I missed. Maybe we can check if the drift is also present in other robotics simulations that use gravity compensation.

Please let me know if anything is unclear and needs clarification.

rickstaa, 2021-12-08:
# Compare different physics engine

Let's quickly try to use a different physics engine while launching the panda simulation. I added [4c7d16536b86102f849ad9d94ff9c4455b543882](https://github.com/rickstaa/franka_ros/commit/4c7d16536b86102f849ad9d94ff9c4455b543882) onto the `develop` branch to do this (see [rickstaa/franka_ros/develop-gazebo-bug-test](https://github.com/rickstaa/franka_ros/commits/develop-gazebo-bug-test)). This commit adds the `physics` argument to the `panda.launch` file.

## ODE

[image]

## DART

[image]

## Bullet

[image]

## Simbody

[image]

## Conclusion

It looks like the drift is less when we use the dart physics engine compared to the ODE physics engine. The simulation does not work in the Simbody and Bullet engines.

rickstaa, 2021-12-08:
@gollth, @marcbone I concluded my investigation into the motion drift; you can check the comments above. I added the plots and figures under a collapsable `<details>` section to remove noise. I summarized my results below:

## Conclusions

- There appears to be a small startup and command delay for the effort coming from Gazebo (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982933376). This delay only seems to be present when no effort control commands are sent. It becomes neglectable when the gravity compensation is enabled and should not be a problem.
-  Gazebo appears to have a slight joint drift when your 7DOF stick is loaded. This drift is likely due to numerical instabilities but is very small and should therefore not influence simulation experiments in which the robot is controlled (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985620934). This drift is removed by increasing the joint damping, but this should only be done if it is kept close to the real robot.
- Both the original long stick and the fully migrated stick have non-zero joint velocities when the Gazebo simulation is started. Altough these velocities are small, I don't think they need to be there. I have not yet found what causes these initial velocities (https://github.com/frankaemika/franka_ros/issues/160#issuecomment-985593607).
- One bug in the current gravity compensation is that the parameters of the `safety_controller` are probably tuned incorrectly (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-988621107).
- Another error in the gravity compensation is that it does not account for the hard joint limits (see https://github.com/frankaemika/franka_ros/issues/160#issuecomment-988621107).
- The panda Gazebo simulation seems to have less drift when using the DART physics engine (https://github.com/frankaemika/franka_ros/issues/160#issuecomment-989173918).

## Next steps

The last two points can be solved by improving the `safety_controller` parameters and/or adding a simple check before the gravity compensation is applied. I will leave this to you since there are many ways to implement this. I think solving this will significantly improve the gravity compensation. 

For the rest of the drift, I noticed that it becomes problematic when the collision geometries are added, but I could not yet find out why this is the case. I think this drift is too big to be caused by the small error found between the KDL gravity component calculation and the actual, but more experimentation needs to be done before we can draw this conclusion. Maybe the drift can be reduced by improving the link, joint and physics engine properties?

Since I have been working on this issue for a long time, I think a fresh pair of eyes could spot what I missed. Maybe we can check if the drift is also present in other robotics simulations that use gravity compensation?

Please let me know if anything is unclear and needs clarification.

rickstaa, 2021-12-08:
> Hello @rickstaa,
> 
> first of all thanks a lot for your great efforts you put into investigation of this bug. I didn't have yet the time to retrace your whole steps but definitely will in the near future.
> 
> Quick comment I noticed right away: I think there is a difference between _Gazebo 9_ & _Gazebo 11_. I tried the example from your fork on my machine (Ubuntu 18.04) and got _different_ results than you (I guess you run 20.04?) after a few time steps:
> 
> ```
> ▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L335 ⟫ panda_joint6: Error:                 0.250258
> ▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L336 ⟫ panda_joint6: Total Effort:          0.000003
> ▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L337 ⟫ panda_joint6: KDL GravComp:          0.250255
> ▏ WARN▕ /gazebo ⟩ FrankaHWSim::writeSim() ⟩ L338 ⟫ panda_joint6: KDL GravComp flipped:  -0.250255
> ```
> 
> Notice the error is _not_ zero. I was trying on ROS Noetic (20.04) and there I get similar values to [those you posted above](https://github.com/frankaemika/franka_ros/issues/160#issuecomment-982682096).
> 
> > I cannot reproduce the error between the `KDL GravComp` and `Total Effort` using your example
> 
> That might also explain why you couldn't reproduce my bugged case. Could you verify that?

@gollth I checked the bug in a singularity container, and I can reproduce it in Gazebo 9. Maybe let's open an issue at https://github.com/osrf/gazebo/issues so that the Gazebo team can backport the fix from Gazebo 11 into Gazebo 9.

```bash
[ WARN] [1639001278.466744774, 0.006000000]: =======================================
[ WARN] [1639001278.466782673, 0.006000000]: panda_joint1: Error:                 0.000000
[ WARN] [1639001278.466802777, 0.006000000]: panda_joint1: Total Effort:          0.249999
[ WARN] [1639001278.466823221, 0.006000000]: panda_joint1: KDL GravComp:          -0.249999
[ WARN] [1639001278.466844545, 0.006000000]: panda_joint1: KDL GravComp flipped:  0.249999
[ WARN] [1639001278.466871029, 0.006000000]: panda_joint1: Gazebo body2Torque:   [0.755914, 0.249999, -0.000000]
[ WARN] [1639001278.466890500, 0.006000000]: panda_joint1: Desired joint command: 0.000000
[ WARN] [1639001278.466902169, 0.006000000]: panda_joint1: Joint command:         -0.249999
[ WARN] [1639001278.466913920, 0.006000000]: panda_joint1: Global Axis: [-0.000000, 1.000000, 0.000000]
[ WARN] [1639001278.466925939, 0.006000000]: panda_joint1: Local Axis:  [0.000000, 1.000000, 0.000000]
[ WARN] [1639001278.466942148, 0.006000000]: panda_joint1: URDF Axis:   [0, 1, 0]
[ WARN] [1639001278.466958541, 0.006000000]: panda_joint1: Joint velocity:        -0.000000
[ WARN] [1639001278.466971492, 0.006000000]: ---------------------------------------
[ WARN] [1639001278.466998382, 0.006000000]: panda_joint2: Error:                 0.000000
[ WARN] [1639001278.467015610, 0.006000000]: panda_joint2: Total Effort:          -0.000000
[ WARN] [1639001278.467027690, 0.006000000]: panda_joint2: KDL GravComp:          0.000000
[ WARN] [1639001278.467044391, 0.006000000]: panda_joint2: KDL GravComp flipped:  -0.000000
[ WARN] [1639001278.467061000, 0.006000000]: panda_joint2: Gazebo body2Torque:   [0.755254, 0.249999, -0.000000]
[ WARN] [1639001278.467076690, 0.006000000]: panda_joint2: Desired joint command: 0.000000
[ WARN] [1639001278.467091885, 0.006000000]: panda_joint2: Joint command:         0.000000
[ WARN] [1639001278.467104446, 0.006000000]: panda_joint2: Global Axis: [-0.000000, -0.000000, 1.000000]
[ WARN] [1639001278.467121542, 0.006000000]: panda_joint2: Local Axis:  [0.000000, 0.000000, 1.000000]
[ WARN] [1639001278.467134261, 0.006000000]: panda_joint2: URDF Axis:   [0, 0, 1]
[ WARN] [1639001278.467150260, 0.006000000]: panda_joint2: Joint velocity:        0.000000
[ WARN] [1639001278.467164855, 0.006000000]: ---------------------------------------
[ WARN] [1639001278.467253730, 0.006000000]: panda_joint3: Error:                 0.000000
[ WARN] [1639001278.467278333, 0.006000000]: panda_joint3: Total Effort:          -0.000000
[ WARN] [1639001278.467294934, 0.006000000]: panda_joint3: KDL GravComp:          0.000000
[ WARN] [1639001278.467313990, 0.006000000]: panda_joint3: KDL GravComp flipped:  -0.000000
[ WARN] [1639001278.467334092, 0.006000000]: panda_joint3: Gazebo body2Torque:   [0.754488, 0.250000, -0.000000]
[ WARN] [1639001278.467350540, 0.006000000]: panda_joint3: Desired joint command: 0.000000
[ WARN] [1639001278.467366407, 0.006000000]: panda_joint3: Joint command:         0.000000
[ WARN] [1639001278.467383865, 0.006000000]: panda_joint3: Global Axis: [-0.000000, -0.000000, 1.000000]
[ WARN] [1639001278.467400347, 0.006000000]: panda_joint3: Local Axis:  [0.000000, 0.000000, 1.000000]
[ WARN] [1639001278.467418461, 0.006000000]: panda_joint3: URDF Axis:   [0, 0, 1]
[ WARN] [1639001278.467437366, 0.006000000]: panda_joint3: Joint velocity:        -0.000000
[ WARN] [1639001278.467451594, 0.006000000]: ---------------------------------------
[ WARN] [1639001278.467554152, 0.006000000]: panda_joint4: Error:                 -0.000000
[ WARN] [1639001278.467579015, 0.006000000]: panda_joint4: Total Effort:          -0.000000
[ WARN] [1639001278.467596277, 0.006000000]: panda_joint4: KDL GravComp:          0.000000
[ WARN] [1639001278.467613028, 0.006000000]: panda_joint4: KDL GravComp flipped:  -0.000000
[ WARN] [1639001278.467630887, 0.006000000]: panda_joint4: Gazebo body2Torque:   [0.753642, 0.250000, -0.000000]
[... cut ...]
[ WARN] [1639001278.468352097, 0.006000000]: panda_joint7: Global Axis: [1.000000, 0.000000, 0.000000]
[ WARN] [1639001278.468367967, 0.006000000]: panda_joint7: Local Axis:  [0.000000, -0.000000, -1.000000]
[ WARN] [1639001278.468382567, 0.006000000]: panda_joint7: URDF Axis:   [0, 0, 1]
[ WARN] [1639001278.468394916, 0.006000000]: panda_joint7: Joint velocity:        -0.000033
[ WARN] [1639001278.468409759, 0.006000000]: ---------------------------------------
[ WARN] [1639001278.468447023, 0.006000000]: TOTAL: 0.250002
```

rickstaa, 2021-12-09:
@gollth I used the singularity recipe below to verify that the bug is present in Gazebo 9. The only result that I get different from you is that Gazebo's effort is `0.0`. I think this is caused by the fact that the results in your table use another axis of rotation than is used in the configuration shown in the figure above. Nevertheless, in my example, Gazebo's torque and joint effort are also incorrect. Maybe you can open an issue on [osrf/gazebo_models](https://github.com/osrf/gazebo_models) repository if you Franka also wants to support ROS melodic?

```bash
[ WARN] [1639081235.413943968, 2.107000000]: panda_joint6: Error:                 0.250003
[ WARN] [1639081235.413970640, 2.107000000]: panda_joint6: Total Effort:          0.000003
[ WARN] [1639081235.414014090, 2.107000000]: panda_joint6: KDL GravComp:          0.250000
[ WARN] [1639081235.414040982, 2.107000000]: panda_joint6: KDL GravComp flipped:  -0.250000
[ WARN] [1639081235.414069299, 2.107000000]: panda_joint6: Gazebo body2Torque:   [0.000007, -0.500000, 0.000003]
[ WARN] [1639081235.414095318, 2.107000000]: panda_joint6: Desired joint command: 0.000000
[ WARN] [1639081235.414120661, 2.107000000]: panda_joint6: Joint command:         0.250000
[ WARN] [1639081235.414149246, 2.107000000]: panda_joint6: Global Axis: [-0.000001, -1.000000, -0.000001]
[ WARN] [1639081235.414176269, 2.107000000]: panda_joint6: Local Axis:  [0.000000, -1.000000, 0.000000]
[ WARN] [1639081235.414204518, 2.107000000]: panda_joint6: URDF Axis:   [0, 0, 1]
[ WARN] [1639081235.414230392, 2.107000000]: panda_joint6: Joint velocity:        -0.000009
```

#### Singularity recipe

```singularity
# Ubuntu 18.04 - Ros melodic Franka gazebo gravity compensation test container
Bootstrap: docker
From: osrf/ros:melodic-desktop-full

%help
    A small container used for testing the problems explained in
    https://github.com/frankaemika/franka_ros/issues/160#issuecomment-961780423.
%labels
    Maintainer: Rick Staa
    Github: https://github.com/frankaemika/franka_ros/issues/160#issuecomment-961780423
    Version: v0.0.0
    Type: Public
%environment
    # Set language options
    export LANG=C.UTF-8 LC_ALL=C.UTF-8

    # Set ROS vars
    export ROS_DISTRO=melodic
%post
    # Install extra system package dependencies
    apt update --fix-missing
    apt install -q -y git

    # Install ros build tools
    apt install -q -y python-catkin-tools

    # Install libfranka from source
    # NOTE: Uncomment if you want to install from source
    apt update
    apt install -q -y \
       cmake \
       libpoco-dev \
       libeigen3-dev
    bash -c "cd / \
       && apt remove -y \"*libfranka*\" \
       && git clone --recursive https://github.com/frankaemika/libfranka.git \
       && cd libfranka \
       && mkdir build \
       && cd build \
       && cmake -DCMAKE_BUILD_TYPE=Release .. \
       && cmake --build ."

    # Build the gazebo gravity compensation test catkin workspace
    mkdir -p /gazebo_gravity_comp_test_ws; cd /gazebo_gravity_comp_test_ws
    git clone -b melodic-devel --recursive https://github.com/rickstaa/gazebo-torque-calculation-bug.git src
    rosdep install --from-paths src --ignore-src --rosdistro melodic -y --skip-keys libfranka
    . /opt/ros/melodic/setup.sh; catkin build -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=/libfranka/build
```

rickstaa, 2021-12-09:
@golth, @marco I just updated my conclusion since I noticed that the drift I removed entirely when the `safety_controller` is disabled in the DART physics engine. I now have a clear picture of what caused this drift and how we can solve it.

rickstaa, 2021-12-10:
@gollth, @marcbone The first part of the solution is implemented in #211.

rickstaa, 2021-12-13:
@gollth, @marcbone I just noticed that I made an error while looking at the safety controller. It is not the safety controller that causes the drift but the addition of the hand. I updated https://github.com/frankaemika/franka_ros/issues/160#issuecomment-989173918 and https://github.com/frankaemika/franka_ros/issues/160#issuecomment-989175979 so that it reflects the right conclusion.

rickstaa, 2021-12-13:
@gollth, @marcbone I think I found the cause of the drift that is experienced when the hand is attached. It looks that in your gravity compensation is not taking into account the physical properties of the attached hand:

https://github.com/frankaemika/franka_ros/blob/cbe64f9e9499dcb192c152f5a6a9473fefc1ff91/franka_gazebo/src/model_kdl.cpp#L277

The currently used `KDL::Chain` starts from `panda_link0` and ends at `panda_link8`. Can you verify that this is indeed the case? I did not create a pull request for this since it would require me to rewrite a significant part of your codebase.

rickstaa, 2021-12-14:
@gollth, @marcbone sorry for the enormous amount of information, but I just noticed I forgot something in my previous conclusion https://github.com/frankaemika/franka_ros/issues/160#issuecomment-992776684. It did not include the fact that although the starting drift is not caused by the `safety_controller` it does cause some unstable behaviour at the soft joint limits. I hope I have given you a good picture of all the problems that are currently present in the gravity compensation. I will leave the further investigation to you now. Please let me know when things are unclear.

gollth, 2021-12-16:
> It looks that your gravity compensation is not taking into account the physical properties of the attached hand:

Yes it does (kind of), directly with the line after your snippets:

https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/model_kdl.cpp#L277-L278

This appends a `load` frame to the end of the chain. This load frame is defined by the mass of the EE (`m_total`) and the CoM (`F_x_Ctotal`) which both come ultimately from the URDF (unless the user sets custom values via ROS parameters):

https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L440-L455

Although you are right, the inertia tensor of this augmented link is still identity

gollth, 2021-12-16:
Thanks for all your investigation @rickstaa

rickstaa, 2021-12-17:
@gollth I created an issue on the [gazebo](https://github.com/osrf/gazebo) repository for the drift difference that is observed between the ODE and DART physics engine (see https://github.com/osrf/gazebo/issues/3149). Did you already create an issue for the ROS melodic torque calculation problem? In my issue, I quickly mention the melodic torque problem but I currently link it to https://github.com/frankaemika/franka_ros/issues/160#issuecomment-987682342.

justagist, 2022-01-19:
Although franka_hw_sim_plugin handles the gravity compensation for the arm, there is no gravity compensation in franka_gripper_sim. This causes the gripper commands to not work correctly when the gripper fingers are not in the same horizontal plane.
I had to completely disable gravity in the simulation to make the gripper work in such configurations.

rickstaa, 2022-01-24:
@gollth, @marcbone do you have any updates on what goes wrong with the gravity compensation when the gripper is attached?

I think using the DART simulator is a good workaround for now when the gripper is not attached. I can successfully train my RL algorithm using the DART simulator as the gravity compensation seems stable. I am unsure what causes the drift when the ODE physics engine is used. I did not have time to dive deeper into the Gazebo codebase, and the gazebo team has not yet responded to [the issue](https://github.com/osrf/gazebo/issues/3149) I opened about this drift. I, therefore, think for now it might be best to merge https://github.com/frankaemika/franka_ros/pull/211 since this gives users the ability to use the DART physics engine when using effort control.

I, however, found one problem when using the DART physics engine. See the section below. Nevertheless, the gravity compensation does not seem to work both in the ODE and DART physics engine, the panda robot keeps drifting, so I think this should be fixed first. 

### Grasp action DART crash

When the gripper is attached and the `franka_gripper/grasp/goal`, `franka_gripper/move/goal` or `franka_gripper/gripper_action/goal` actions are used the simulator sometimes crashes with the following error message:

<details>
<summary>Show error message</summary>

Error [BoxedLcpConstraintSolver.cpp:291] [BoxedLcpConstraintSolver] The solution of LCP includes NAN values: -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan -nan - [... cut ...] emika/franka_ros/pull/173#issuecomment-979223037. However, I did not yet investigate since the gravity compensator for when the gripper is attached is not working yet, which might also cause the crash. However, this hypothesis is strengthened since it does not seem to happen when I stabilize the panda arm by putting it down on the ground (see figure below). I know that getting the gripper to work in simulation is not easy, but I just wanted to make you aware of all the problems I found.

[image]

rickstaa, 2022-01-24:
> Although franka_hw_sim_plugin handles the gravity compensation for the arm, there is no gravity compensation in franka_gripper_sim. This causes the gripper commands to not work correctly when the [...]

@justagist Thanks for your comment! @marcbone I think this is closely related to my comment in https://github.com/frankaemika/franka_ros/pull/173#issuecomment-979223037. The gripper PID gains might be easier to tune when the gravity of the fingers is also compensated! I however do not think that this is the major cause of the drift we observe when the hand is attached when using the DART physics engine (see figure below).

[image]

gollth, 2022-03-11:
For reference, the movement of the joints should now be gone also in Melodic with the new feature from @rickstaa & @marcbone: https://github.com/frankaemika/franka_ros/pull/181#issuecomment-1064120809 (although it does only mask the underlying real problem in Gazebo).

@rickstaa what do you think, shall we close this ticket or do you want to keep it open for reference?

rickstaa, 2022-03-18:
@gollth Although, I agree with you that the problem as it was originally described by @sallimperator has been fixed the underlying problem still is present. I however think we can close #160 and continue the discussion on this problem in #233.

rickstaa, 2022-03-18:
> > Although franka_hw_sim_plugin handles the gravity compensation for the arm, there is no gravity compensation in franka_gripper_sim. This causes the gripper commands to not work correctly when the gripper fingers are not in the same horizontal plane. I had to completely disable gravity in the simulation to make the gripper work in such configurations.
> 
> @justagist Thanks for your comment! @marcbone I think this is closely related to my comment in [#173 (comment)](https://github.com/frankaemika/franka_ros/pull/173#issuecomment-979223037). The gripper PID gains might be easier to tune when the gravity of the fingers is also compensated! I however do not think that this is the major cause of the drift we observe when the hand is attached when using the DART physics engine (see figure below).
> 
> See gif

I just double-checked and for the gripper joints the gravity is being compensated:

https://github.com/frankaemika/franka_ros/blob/30dc6b8cc3627a9687636dc8811c3409b58027fc/franka_gazebo/src/franka_hw_sim.cpp#L365-L377

Could still be that the component of the gravity of the finger and hand on the other arm joints is not being handled correctly like @justagist states.

rickstaa, 2023-11-20:
I just caught a mistake in my previous message—I mixed up some details. Titans benchmark, with its single test stream running for 60 seconds, probably isn't the culprit for the issues. However, it's possible that the problem arises from the concurrent operation of both the titan-node binary and the go-live peer docker container. When high-resolution streams kick in, the GPU memory maxes out rapidly. The quick pace of this occurrence might be why it doesn't immediately show up on the Grafana dashboard.