frankarobotics/franka_ros #167: Jerkyy movements when controlling the panda using the 'ros_control' 'joint_position/JointPositionController'

rickstaa, 2021-09-03:
I am trying to control the real Panda robot using the [joint position motion generator](https://frankaemika.github.io/docs/libfranka.html#realtime-commands). I currently use the [joint_position/JointPositionController](https://github.com/ros-controls/ros_controllers/tree/melodic-devel/position_controllers/include/position_controllers) controller that is found in the [ros-controls/ros_control](https://github.com/ros-controls/ros_control) package.

Unfortunately, the control is very unstable. Although, I am able to send a control commands, the robot oscillates around the set point (see rviz visualization GIF below). I inspected the code of the [franka_ros](https://github.com/frankaemika/franka_ros/), [libfranka](https://github.com/frankaemika/libfranka)  and [ros_Control](https://github.com/ros-controls/ros_control) packages and it looks like the ROS controller command is correctly forwarded to the robot class where it is used in the motion planner callback. My assumption therefore is that the Joint position motion generator can not handle non-smooth set-points.

It looks like my assumption is correct since sending a smooth signal, like is done in the examples you provide (i.e. https://github.com/frankaemika/franka_ros/blob/noetic-devel/franka_example_controllers/src/joint_position_example_controller.cpp and https://github.com/frankaemika/libfranka/blob/master/examples/generate_joint_position_motion.cpp) solves the oscillation issue.This is contrary to what I would have expected, namely that the motion generator would generate the motion to get the robot to a given position set point. 

Is my assumption correct and can we only send smooth paths to the Joint Position motion generator, or is this a bug within my current implementation?

[image]

#### Steps to reproduce the issue

1. Clone [this example repository](https://github.com/rickstaa/real-panda-control-examples/tree/noetic-devel-ws).
2. Checkout the `noetic-devel-ws` branch.
3. Build the catkin workspace.
4. Run the `real_panda_position_control.launch` file.
5. Try to control the robot by publishing on the 'panda_arm_joint7_position_controller/command' topic or by using the rqt GUI.
6. See the oscillations take place.

fe-support, 2023-06-20:
Closing due to inactivity. Please open this git issue again if the topic still exists.

rickstaa, 2023-06-20:
> Closing due to inactivity. Please open this git issue again if the topic still exists.

@fe-support, closing this issue requires verifying whether you can reproduce the problem. The last time I checked (4 months ago this issue still was present. Maybe we can re-open this issue until @frankaemika verifies it is fixed?