frankarobotics/franka_ros #168: connected_to option in panda_gazebo.xacro

LucasJoseph, 2021-09-06:
It would be nice to have the `connected_to` argument in the panda_gazebo.xacro, similarly to panda_arm.xacro.

gollth, 2021-09-27:
Sure, we can do that. May I ask how do you use this arg? Do you include the `panda_arm.xacro` in another more complex URDF?

LucasJoseph, 2021-11-09:
Yeah, my robots are fixed on a table at different height and I include several `panda_arm.xacro` in my URDF.

gollth, 2021-11-09:
Alternatively you can also connect the robot in the Xacro where you instantiate the `panda_arm.xacro` macro, right? Something like:

```xml
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="myrobot">
  <xacro:include filename="$(find franka_description)/robots/panda_gazebo.xacro" />

  <xacro:panda_arm arm_id="panda" />

  <link name="YOUR_TABLE_LINK" />

  <joint name="table_joint" type="fixed">
    <origin xyz="0 0 TABLE_HEIGHT" rpy="0 0 0" />
    <parent link="YOUR_TABLE_LINK" />
    <child  link="panda_link0" />
  </joint>

<robot>
```

Would that also work for you?

LucasJoseph, 2021-11-16:
Well, I could and that's what I did for now, but since the option is available in the `panda_arm.xacro` I wanted to be able to use it also with `panda_gazebo.xacro`

gollth, 2022-01-05:
Make sense to have the same interface in both files. Added in bd2bfd8. Thanks for the input

closed 2022-01-05 by gollth