frankarobotics/franka_ros #172: [bug] Gazebo franka_gripper_sim GripperCommand action doesn't respect gripper width.

rickstaa, 2021-09-27:
In the current version, when working with the gazebo simulation, the gripper width that is sent to the `/panda/franka_gripper/gripper_action/goal` topic is not coming through. This is because the following code:

https://github.com/frankaemika/franka_ros/blob/054af531f43f3fc1ddb439cc4102a41f87a8e0da/franka_gazebo/src/franka_gripper_sim.cpp#L436

Due to this, the desired gripper with now becomes `0` when the gripper width is smaller than the current gripper width and `0.08` when it is bigger than the current gripper width. Consequently, users can now only open/close the gripper. 

I created #171 to address this problem.

#### Steps to reproduce

gollth, 2021-09-28:
Hi @rickstaa,

thanks for working on this and the whole moveit integration. The problem with this approach is the way how the grasping works. You probably already noticed the statemachine logic inside `franka_gripper_sim`. The core transitioning mechanic happens here: https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gazebo/src/franka_gripper_sim.cpp#L147-L191

Right now, the target width `w_d` (which originates from the goal of the GripperCommand message) is checked each cycle to be within certain tolerances. If it is, the statemachine will switch eventually to `State::IDLE`, not `State::HOLDING` (in which forces are exerted during static contact with an object). 

With your solution, as soon as the fingers reach the desired width +/- threshold they would stop and don't hold the object tight!

Our original solution with the ternary operator was to specify widths _beyond_ the desired width (i.e. 0 or 8cm) and use the width from the goal only after the grasp was established to check if the action succeeded or failed: 

https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gazebo/src/franka_gripper_sim.cpp#L457-L469

Please correct me if this is not the indented way how this gripper command action is supposed to work, though.

rickstaa, 2021-09-30:
@gollth Thanks again for your quick and concise answer! I apologize in advance for the lengthy discussion below. I tried to be as brief as possible, but it amounts to a long read.

> With your solution, as soon as the fingers reach the desired width +/- threshold they would stop and don't hold the object tight!

I tried debugging my code, but I could not reproduce this behaviour. I can not find any code that transfers the state machine to `IDLE` state when the `FrankaGripperSim::onGripperActionGoal` initially pushes it to `GRASPING` state (see the gif below). 

https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gazebo/src/franka_gripper_sim.cpp#L149-L168

I, however, discovered two other problems that are present in my solution.

### Problems 

#### Wrong gripper PID gains

In the current form, it looks like the PID gains of the gripper prevent the velocity from becoming lower than the `kGraspRestingThreshold` (see gif below). As a result, the state machine often gets stuck in the `GRAPSING` state. This can be fixed by lowering the derivative gain to 0.0. Doing this, however, can lead to overshoots. If we want to use a solution like #173 the gripper gains, therefore, must be re-tuned first.

In the old codebase, this behaviour goes unnoticed, since the gripper only moves between the open and closed positions due to the ternary expression. In these positions, the gripper is in contact with the external world, which damps out the instabilities caused by the high derivative gain.

#### Moveit doesn't send any force 

I noticed that Moveit currently doesn't allow users to set the 
`max_effort` value of the [GripperCommand](https://docs.ros.org/en/noetic/api/control_msgs/html/msg/GripperCommand.html) message (see the [move_group code API](https://docs.ros.org/en/noetic/api/moveit_commander/html/move__group_8py_source.html#l00199)) when they use the `move_group` class. As a result, the `command.max_effort` is set to 0, which means that the robot will apply no force in the `HOLDING` state. Therefore, in the current form, the robot is likely inadequate to hold heavy objects in the simulation.

This is not a problem with your API but a feature request that should be opened on the MoveIt repository. 

#### Why did I open this issue

I opened this issue to make sure the gripper behaviour in the simulation is as close as possible to that of the real robot. If I remember it correctly, I could set a specific width and grasp with the actual robot on the real robot. Giving users the ability to specify a gripper width and grasp objects would bring the simulation closer to the real robot.

<details>
<summary>I quickly reviewed the codebase of the [franka_gripper](https://github.com/frankaemika/franka_ros/tree/develop/franka_gripper) package, and it looks like the `GripperAction` is handling slightly different there.<summary>

To my current understanding, in the `franka_gripper` node when the desired width is smaller than the current width, the `move` action is executed, and the gripper is closed until it is within the `kSamePositionThreshold`. 

https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gripper/src/franka_gripper.cpp#L52

In this case, the gripper keeps moving when an object is encountered, applying the force needed to lift the object. The ' grasp ' action is executed when the gripper width is more significant than the current width.

https://github.com/frankaemika/franka_ros/blob/1922fc9d24e37b2a8226c522e736eaa9a65d66ca/franka_gripper/src/franka_gripper.cpp#L54

I can't look into the FCI, but I think this action tries to apply the requested force. 

</details>

gollth, 2021-10-01:
Hi @rickstaa,

> tried debugging my code, but I could not reproduce this behaviour. I can not find any code that transfers the state machine to `IDLE` state when the `FrankaGripperSim::onGripperActionGoal` initially pushes it to `GRASPING` state (see the gif below).

Yes your are absolutely right, I was confusing things in my mind. Sorry about that. So #173 should work and bring `franka_gripper_sim` and `franka_gripper` more closely together. We will discuss over there a strategy how to merge that.

Regarding the problem you have with getting stuck in `GRASPING` state. I agree that this has to do gazebo gains. With the default thresholds the fingers seem to oscillate around and over the `kGraspRestingThreshold`. Luckily you can change them via ROS parameter ;) I fiddled around a bit with the threshold and the gains and the following works for me Plan & Execute in RViz Motion Planning:

```yaml
# File: franka_gazebo/config/franka_hw_sim.yaml
franka_gripper:
  type:    franka_gazebo/FrankaGripperSim
  arm_id:  $(arg arm_id)

  finger1:
    gains: { p: 1000, i: 0, d: 0 }

  finger2:
    gains: { p: 1000, i: 0, d: 0 }

  grasp:
    resting_threshold: 0.05
    consecutive_samples: 10
```

(I'm think we should even move the `resting_threshold` and `consecutive_samples` parameter outside of `grasp` namespace, since they are also relevant for `gripper_command`)

[image]

So I suggest we discuss in #173 about how to merge that and hopefully you can continue with these gains with the MoveIt update. How does that sound?

rickstaa, 2021-11-23:
As stated in https://github.com/frankaemika/franka_ros/pull/180#issuecomment-976605845 I think the behaviour that was pointed out by @rhaschke in #173 (comment) would be the desired behaviour to solve this issue.

> IMO, the gripper action should realize position-control as long as measured efforts are smaller than max_effort. Otherwise, it should switch to force-control and maintain the given (grasp) force.

rickstaa, 2021-11-23:
@gollth, @marcbone I performed some grasping tests using the new development branch, and I think you are close to solving this issue. I, however, found the following problems.

### Move action

The move action gives some peculiar behaviour when moving the gripper beyond the `stone` object boundary. This is not too problematic since the move action is not meant to grasp objects. I, however, wanted to make you aware of this issue since it might confuse people when they try out the tutorial.

In the gif below, I first close the gripper using the `move` action. When I do this, the `stone` moves up and down relative to the gripper while the gripper is not moving. This, is probably caused by the properties of your `stone` object since when I use one of my objects that also work with other robots; The behaviour is not there (see gif2). In both cases, as expected, the object slides out of the gripper.

[image]

[image]

**My cube:**

```sdf
<?xml version='1.0'?>
<sdf version='1.7'>
  <model name='cube'>
    <link name='link'>
      <inertial>
        <pose>0.03 0.03 0.03 0 -0 0</pose>
        <!--NOTE: Uses Polyethylene density to calculate the mass
        see https://en.wikipedia.org/wiki/Polyethylene
        -->
        <mass>0.20735999999999996</mass>
        <inertia>
          <ixx>0.00012442</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.00012442</iyy>
          <iyz>0</iyz>
          <izz>0.00012442</izz>
        </inertia>
      </inertial>
      <visual name='visual'>
        <pose>0.03 0.03 0.03 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.06 0.06 0.06</size>
          </box>
        </geometry>
        <material>
          <script>
            <uri>file://media/materials/scripts/gazebo.material</uri>
            <name>Gazebo/DarkGrey</name>
          </script>
          <shader type='pixel'/>
        </material>
      </visual>
      <collision name='collision'>
        <pose>0.03 0.03 0.03 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.06 0.06 0.06</size>
          </box>
        </geometry>
        <surface>
          <!--NOTE: Used dynamic friction of clean dry plastic cube on a plastic surface
          (i.e. Polyethylene) see https://www.engineeringtoolbox.com/friction-coefficients-d_778.html
          -->
          <friction>
            <ode>
              <mu>0.2</mu>
              <mu2>0.2</mu2>
            </ode>
          </friction>
          <contact>
            <collide_bitmask>1</collide_bitmask>
            <ode>
              <kp>1e+06</kp>
              <kd>1e+06</kd>
              <max_vel>1</max_vel>
              <min_depth>0.002</min_depth>
            </ode>
            <bullet>
              <kp>1e+06</kp>
              <kd>1e+06</kd>
            </bullet>
          </contact>
        </surface>
      </collision>
    </link>
  </model>
</sdf>
````

### Gripper action

In the current form, the gripper action still is unable to grasp objects (see gif).

[image]

rickstaa, 2021-11-23:
I quickly looked into the move action problem and it is indeed caused by the definition of the `stone` and `pick_tray`. The inertias and centre of masses of these objects look very strange compared to my grasping object. I think you should improve these.

[image]

gollth, 2021-11-23:
> In the current form, the gripper action still is unable to grasp objects (see gif).

This is kind of expected, when you pass a gripper command position of 0.0. When you check the result of the action or the [documentation of franka_gripper](https://frankaemika.github.io/docs/franka_ros.html#franka-gripper) you will find out why. For the same reason, grasping works in your example above

gollth, 2021-11-23:
> The inertias and COM of these objects look very strange compared to my grasping object. I think you could improve these by using real material properties.

Thanks @rickstaa for finding that out. Would you mind a PR?

rickstaa, 2021-11-23:
@gollth I'm happy to create a pull request. I however have to know which materials you want to use. Maybe stone for the stone and wood for the trays.

gollth, 2021-11-23:
Stone made of stone material, sounds legit =D

rickstaa, 2021-11-23:
@gollth I created draf in #206. There however are still some problems (see the pull request).

rickstaa, 2021-11-24:
For future reference, part of the discussion of this topic is found in #173.

rickstaa, 2021-11-25:
To summarize. My bug report is related to closely related issues:

- First, it looks like the current implementation of the `gripper_action` and thus `grasp` service does not work. The State Machine does not transition into the HOLDING state when high I and D gains are used since the velocities fluctuate too much. This is solved by lowering the I and D gains. The downside is that the gains are now not high enough to control the fingers when the hand is not horizontal.

- Secondly, it is about the implementation of the `gripper_action` service. In the current form, MoveIt can not both grasp and move the gripper (see https://github.com/frankaemika/franka_ros/pull/173#issuecomment-977996272).

rickstaa, 2022-03-18:
@gollth, @marcbone I updated the title to better represent the problems that are still present with the gripper in simulation.

rickstaa, 2023-11-20:
Certainly! Here's an improved version of the bug report:

---

## Bug Report: Gripper Simulation Issues

### Feedback and Investigation

I received user feedback regarding issues with the gripper simulation, as reported in [GitHub issue #33](https://github.com/rickstaa/panda-gazebo/issues/33#issuecomment-1586089869). To address these concerns, I investigated whether the problems had been resolved upstream. Below are my findings:

### Current Gripper Simulation Status

The simulated gripper has seen improvements since the initial report. However, two critical bugs persist:

1. **Gripper PID Gains:**
   - Gripper PID gains need improvement for the `/franka_gripper/gripper_action/goal` Action to set the gripper width accurately when the gripper is vertical.

2. **Gripper Action Behavior:**
   - The `/franka_gripper/gripper_action/goal` action should be enhanced. It should enable position control when measured efforts are below `max_effort` and switch to force control; otherwise, maintain the specified grasp force.

### Proposed Improvements

Addressing these issues will enhance the usability of the Panda simulation for RL research. Specifically, it allows users to specify a gripper width with a max force, enabling the gripper to apply that force to encountered objects.

### Gripper Simulation Investigation

#### Overview

| Action                              | Horizontal Move | Vertical Move | Grasp |
|-------------------------------------|-----------------|---------------|-------|
| `/franka_gripper/move/goal`         | ✅               | ✅             | ❌     |
| `/franka_gripper/gripper_action/goal` | ✅               | ❌             | ❌     |
| `/franka_gripper/grasp/goal`        | ❌               | ❌             | ✅     |

#### Move Action Investigation

**Move Action:**
- Successfully moves the gripper to a specific width using `/franka_gripper/move/goal`.

**Grasp Action:**
- Cannot be used to grasp objects because both gripper heads stop when encountering an object, leaving space between the gripper and the target.

[image]

#### Gripper Action Investigation

**Move Action:**
- `/franka_gripper/gripper_action/goal` can set a gripper width by setting `max_effort` to zero.

**Grasp Action:**
- Still cannot be used to grasp an object. The behaviour needs clarification and improvement. Suggested behaviour in [comment](https://github.com/frankaemika/franka_ros/pull/173#issuecomment-966255718) by @rhaschke is considered desirable.
- Gripper action should achieve position control if efforts are below `max_effort`. Otherwise, it should switch to force control, maintaining the specified grasp force.
- Additionally, the Action fails when the gripper is vertical due to incorrect PID gains ([GitHub issue #209](https://github.com/frankaemika/franka_ros/pull/209#issuecomment-1008713174)).

#### Grasp Action

The `/franka_gripper/grasp/goal` cannot move the gripper to a specific width. This is as expected.

**Grasp**

The /franka_gripper/grasp/goal action successfully grasps objects when the correct gripper width is set or the inner epsilon is configured to match the gripper width.

rickstaa, 2023-11-20:
@gollth, following my investigation, I've revised the issue body to include my latest proposed improvements. Feel free to tag me if you have any questions.