frankarobotics/franka_ros #193: Requirements for panda_moveit_config release

rhaschke, 2021-11-07:
The new release of https://github.com/ros-planning/panda_moveit_config is blocked by a few things in `franka_ros`:
- [ ] #189
- [ ] #183
- [ ] #187
- [ ] #190
- [ ] #186

Nice to have:
- [ ] #191
- [ ] Allow to use different hardware transmissions in Gazebo than the default effort transmission
  - [ ] #157
  - [ ] #181
- [ ] #173

Optional:
- [ ] #192

rickstaa, 2021-11-09:
@rhaschke Thanks for collecting all our issues into an overview. One small suggestion can you change

> GripperCommand doesn't signal convergence (but movements are correctly executed) 

to 

> GripperCommand doesn't signal convergence (but movements are correctly executed) (see #173)

The problem is explained in detail in that pull request. Further, another related issue that should be mentioned is that #146, which was merged to the devel branch, broke the gripper actions (see #179).

gollth, 2021-11-10:
For the record, @rhaschke shall we change PR #173 to Issue [#172](https://github.com/frankaemika/franka_ros/issues/172) on the checklist? (as discussed [here](https://github.com/frankaemika/franka_ros/pull/173#issuecomment-965023144))

rickstaa, 2022-03-18:
>A few things block the new release of https://github.com/ros-planning/panda_moveit_configrankaemika/franka_ros/pull/191)
> * [ ]  Allow to use different hardware transmissions in Gazebo than the default effort transmission
>   
>   * [ ]  [Adds the 'transmission' arg to the 'panda_arm.urdf.xacro' file #157](https://github.com/frankaemika/franka_ros/pull/157)
>   * [ ]  [Adds position and velocity motion generators to FrankaHWSim #181](https://github.com/frankaemika/franka_ros/pull/181)

@rhaschke Just for tracking purposes. The motion generators have been implemented in https://github.com/frankaemika/franka_ros/commit/2514ae1990d51ca808e6c49d22d24ae7a99db532.