frankarobotics/franka_ros #201: [feature_request] Decrease joint trajectory controller goal tollerance

rickstaa, 2021-11-16:
As mentioned in https://github.com/frankaemika/franka_ros/commit/e899240365797de6d53c222d6512b74c3a40268e#commitcomment-60095175 the current goal tolerance of the JointTrajctory controller is quite big. I think making it smaller (~`0.005`) would create more accurate behavoir:

https://github.com/frankaemika/franka_ros/blob/e899240365797de6d53c222d6512b74c3a40268e/franka_control/config/default_controllers.yaml#L13-L19

https://github.com/frankaemika/franka_ros/blob/e899240365797de6d53c222d6512b74c3a40268e/franka_control/config/default_controllers.yaml#L41-L47

fe-support, 2023-06-20:
Closing due to inactivity. Please open this git issue again if the topic still exists.

marcbone, 2023-06-21:
yeah we shouldnt have closed this. Nobody really investigated this so I am gonna reopen it. Also its not 5 cm but 0.05 radian :sweat_smile: