frankarobotics/franka_ros #207: [feature_request] Add `tau_grav_hat` to the `franka_state_controller/franka_states`

rickstaa, 2021-11-25:
For one of my projects, I was looking for an easy way to retrieve the torque that is applied by Gravity on the panda_joints. I know that I can retrieve this by using the [getGravity](https://github.com/rickstaa/franka_ros/blob/5341770c6db24cdf5acf1bd87e832cf9f60a1b5c/franka_hw/include/franka_hw/franka_model_interface.h#L122) method of the [franka_hw::FrankaModelInterface](https://github.com/rickstaa/franka_ros/blob/kinetic-devel/franka_hw/include/franka_hw/franka_model_interface.h), but I think it would be a significant improvement if this parameter was included in the `franka_states` that is published by the `franka_state_controller` or in a separate topic like is done with the `F_ext` parameter.

https://github.com/rickstaa/franka_ros/blob/5341770c6db24cdf5acf1bd87e832cf9f60a1b5c/franka_hw/include/franka_hw/franka_model_interface.h#L122
 
I did not create an example pull request for this since there are several ways to implement this.

rickstaa, 2021-11-25:
People can also infer these torques when #198 is merged.

rickstaa, 2021-11-26:
The real problem is that I want to check whether the torque send by the controller has been achieved before continuing with my training script. In the simulation, this is currently not possible without having access to the gravity torques and external torques. This is because `tau_J` contains the control commands, the external torques and the gravity torques. I, therefore, first have to subtract the gravity and external torques to check if the torque command is applied. This should not be a problem in the simulation since I think the effort command is applied instantly. I could not find documentation on this, but this looks to be the case from the ODE code. On the real robot, this effort ramp delay should also be small, right?

rickstaa, 2021-11-26:
From looking at the [documentation](http://gazebosim.org/tutorials?tut=apply_force_torque&cat=tools_utilities), [this question](https://answers.gazebosim.org/question/26675/accuracy-of-gazebo-simulation-of-arm-vs-real-hardware/), [this question](https://answers.gazebosim.org//question/17152/new-gazebo-plugin-for-ros-control-to-simulate-control-packet-latency-robot_hw_sim_latency/) and the gazebo codebase, it looks like there is no native way to simulate temporal control effects. As a result, requested torques will be applied [instantly](https://github.com/osrf/gazebo). As a result, when we want to simulate latency to communication between ros_control and gazebo, we have to add the logic similar to [Ericsson/robot_hw_sim_latency](https://github.com/Ericsson/robot_hw_sim_latency) package. I created https://answers.gazebosim.org/question/27608/are-temporal-control-effects-simulated-in-gazebo/ to see if things have changed compared to 2017, but I will close this issue since I expect my conclusion to be correct.

gollth, 2021-12-07:
Including it in the State Interface will not be the best option, since this data is not published via the FCI and libfranka. We can only add fields into the `franka_states` which come directly from the robot.

Providing a topic seems possible, but I wonder how useful that would be. Most often you need this data in realtime for control, not?

What speaks against the "ros_control"-way of doing this, i.e. claiming the model interface and getting the gravity term via `getGravity()`?

rickstaa, 2021-12-08:
@gollth Ah, if you're not planning to implement this in the FCI, then adding it to the franka_states indeed does not make sense.

There is nothing wrong with using the `getGravity()` function. I, however, thought publishing these gravity torques on a rostopic would improve user experience. By doing so, users who use the high-level franka_ros package can access this information without interacting with the [franka_model_interface](https://github.com/rickstaa/franka_ros/blob/5341770c6db24cdf5acf1bd87e832cf9f60a1b5c/franka_hw/include/franka_hw/franka_model_interface.h) through C++.

gollth, 2021-12-27:
Would definitely improve user experience but at the cost of diverging interfaces between real robot and simulation I'd say this is not worth it, especially if you can already get this information via the model interface otherwise. Feel free to reopen if new insights arise

closed 2021-12-27 by gollth

rickstaa, 2021-12-27:
No problem! I agree with your conclusion.