frankarobotics/franka_ros #215: [docs] Explenation to CartesianImpedanceExampleController

peetCreative, 2021-12-22:
Hey,
I'm working  on a project where I build something similar to the cartesian impedance controller. So I started to try to understand how this example is working. However I have a hard time understanding the physics and math behind the controller just from looking at the code. I'm just in the process of trying to understand the equations using [dimensional analysis](https://en.wikipedia.org/wiki/Dimensional_analysis).
Can you offer some more in depth explanations of how the math and physics of the Controller is working? A link would help as well.
I could imagine contributing some documentation for this controller, because I think it is really useful for a lot of applications.

So basically I have a hard time to understand how the Torque dimension in tau_task and tau_nullspace (M L<sup>2</sup>T<sup>-2</sup>) is composed [here](https://github.com/frankaemika/franka_ros/blob/8efef6235b969c68ec1f8d5bc85026e4bca9cf19/franka_example_controllers/src/cartesian_impedance_example_controller.cpp#L173). 

Which dimensions and units do the variables: cartesian translational/roational stiffness/damping and (jointspace) nullspace stiffness/damping have. 
[Stiffness](https://en.wikipedia.org/wiki/Stiffness) is translational: M L<sup>2</sup>T<sup>−2</sup> (Unit: N/m)  or rotational: M L<sup>1</sup>T<sup>−2</sup> (Unit: N/rad).
The [damping variables](https://github.com/frankaemika/franka_ros/blob/8efef6235b969c68ec1f8d5bc85026e4bca9cf19/franka_example_controllers/src/cartesian_impedance_example_controller.cpp#L224) are calculated depending the stiffness: `damping = 2* sqrt(stiffness)`.  The comment says `damping ration = 1`, which may refer to [damping](https://en.wikipedia.org/wiki/Damping). However I'm missing the mass here. So where does this equation comes from?
And in consequence which dimensions have the different damping variables? 

Futher just for sanity check PD control in the commend means: Proportional and differential control. The proportional part is `stiffness * cartesian_error`/ `stiffness * joint_error` and differential is `damping * jacobian * joint_velocity`/`damping * joint_velocity` ? However for `tau_task`  booth components are substracted as in `tau_nullspace`  the proportional part is added and the differential is substracted?  I tested it and it is correct like this, but I don't get why.

Further I wonder for `tau_nullspace` about the first factor: Here transposed pseudo inverted Matrix is multiplied with the transposed jacobian, so we get an 7x7-Matrix, which should be the identity matrix by [definition](https://en.wikipedia.org/wiki/Inverse_element#Matrices)(not sure I read/remember this correctly), or? So the more or less the first factor get's zero. So what does it mean to create the pseudo inverse of the jacobian for a q? However the first factor must be L<sup>2</sup>. Which makes sense if we multiply the jacobian with another jacobian.

Further for applying new stiffness and equilibrium poses, the controller uses a digital infinit [low-pass filter](https://en.wikipedia.org/wiki/Low-pass_filter). The filter parameter &alpha; is defined `&alpha; = dt/RC-dt` with an &alpha; of 0.005. Is this a random number or maybe dependend on the 1kHz control loop frequency?

I'm learning a lot in the process. Thank you for your patience!

Maverobot, 2023-02-08:
For the start, I would recommend you to read the paper:

* [Cartesian  Impedance  Control  of  Redundant  Robots:  RecentResults  with  the  DLR-Light-Weight-Arms](http://www.informatik.uni-bremen.de/agebv2/downloads/published/albuschaeffericra03.pdf)

It should give you a good overview of the maths regarding Cartesian impedance controller.

Yitian-Xian, 2023-05-08:
@Maverobot Hello, may I know why there is no gravity term when computing the tau_d? This term is added by tau_d << tau_task + tau_nullspace + coriolis, which are from Cartesian PD control, nullspace PD control, and the coriolis effect. Could you please explain that? Thank you.

Maverobot, 2023-05-08:
Hi @Yitian-Xian, the gravity term can be considered as part of tau_task.

Cheers,
Zheng

Yitian-Xian, 2023-05-08:
@Maverobot Hi, thank you for your answer. However, I find it very hard to understand. First of all, in your recommended paper, from Equation 10, we can see that there is a gravity term over there. In addition, what if the current pose is equal to the desired pose, it seems that in this situation tau_task would be zero. 

I am very confused. Could you please explain it further? Thank you very much!!