frankarobotics/franka_ros #218: Gazebo: Failed to create robot simulation interface loader

hennking98, 2022-01-10:
Hello,
I am currently following the gazebo tutorials in the Franka Control Interface documentation.
I installed franka-ros from the ROS repositories with the following command:

`sudo apt install ros-noetic-libfranka ros-noetic-franka-ros`

The franka-gazebo package was missing so I installed myself.
When I now try to run:

`roslaunch franka_gazebo panda.launch`

I get the following error:

> [ INFO] [1641851266.573617743]: gazebo_ros_control plugin is waiting for model URDF in parameter [/panda/robot_description] on the ROS param server.
> [FATAL] [1641851266.676289907]: Failed to create robot simulation interface loader: Could not find library corresponding to plugin franka_gazebo/FrankaHWSim. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

Thank you for your help!

marcbone, 2022-01-11:
Seems like an error on our side.

As a quick fix I would suggest you to remove franka-ros completely with
`sudo apt remove ros-noetic-franka-ros`
and clone the github repo and build everything from source.

mnksnz, 2022-02-10:
Hi, 
I noticed the exact same error also with Melodic. But the source install seems to quickly fix this problem.

gollth, 2022-02-18:
Probably related to https://github.com/frankaemika/franka_ros/issues/175. If so, this is already fixed on `develop`. Once we release the next version, this will be fixed on apt as well

gollth, 2022-03-03:
Hi @hennking98, we recently released franka_ros 0.8.2 which contains the fix for your issue. So now also `sudo apt update && sudo apt upgrade` should fix your issue. The new version is already on Noetic, for Melodic it might take a while to go to upstream.

If problem persists, feel free  to reopen

closed 2022-03-03 by gollth