frankarobotics/franka_ros #220: How to control the robot in gazebo through joint position?

Yitian-Xian, 2022-01-14:
Hello, can anyone tell me how to configure the robot so that I can control each joint position? Currently it seems that it only supports cartesian and joint effort conrtol. Thanks in advance!

rickstaa, 2022-02-02:
@Yitian-Xian The real Panda robot supports position and velocity control through motion generators (see [the franka_ros documentation](https://frankaemika.github.io/docs/franka_ros.html?highlight=motion%20generators#writing-your-own-controller)). For the simulated robot, you have to merge the pull request you already found https://github.com/frankaemika/franka_ros/pull/181. The control behaviour using the motion generators on the real robot is similar to the explanation I gave in https://github.com/frankaemika/franka_ros/pull/181#issuecomment-1019301350.

gollth, 2022-02-03:
@rickstaa thanks for the quick answer! We haven't had the chance to integrate & test https://github.com/frankaemika/franka_ros/pull/181 to be release it officially, but since more and more people request this feature, it will climb in the priority list for sure.

Until then, @Yitian-Xian feel free to fork the repo and use the PR @rickstaa provided

closed 2022-02-03 by gollth

Yitian-Xian, 2022-03-19:
Cool! Thank you for your support! I am able to use now!

rickstaa, 2022-03-19:
@Yitian-Xian As of 2514ae1990d51ca808e6c49d22d24ae7a99db532 there is no need to use my fork anymore. These motion generators are now merged into the `develop` branch.