frankarobotics/franka_ros #222: Error when running franka_control.launch

venkateshsripada, 2022-01-28:
Hello,

I am using 
ubuntu: 16.04; 
ros version: kinetic
FCI version (robot system version): 2.1.1
libfranka version: 0.5.0
franka_ros version:0.6.0
kernel: 4.13.13-rt5

I am trying to run the franka robot in real time.

In libfranka examples, the `generate_catesian_pose_motion` moves the robot 

But when I run `roslaunch franka_control franka_control.launch` I get the following error  

... logging to /home/venkatesh/.ros/log/3b3be3d6-8065-11ec-8dcd-3ca82af3b35c/roslaunch-uolncnrfranka3-17595.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uolncnrfranka3:44439/

SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_gripper/default_grasp_epsilon/inner: 0.005
 * /franka_gripper/default_grasp_epsilon/outer: 0.005
 * /franka_gripper/default_speed: 0.1
 * /franka_gripper/joint_names: ['panda_finger_jo...
 * /franka_gripper/publish_rate: 30
 * /franka_gripper/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 1000
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 1000
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /position_joint_trajectory_controller/constraints/goal_time: 0.5
 * /position_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /position_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /position_joint_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    franka_control (franka_control/franka_control_node)
    franka_gripper (franka_gripper/franka_gripper_node)
    joint_state_desired_publisher (joint_state_publisher/joint_state_publisher)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    state_controller_spawner (controller_manager/spawner)

[... 252 lines cut ...]
7f4c6e50e000-7f4c6e50f000 rw-p 00000000 00:00 0 
7ffe00844000-7ffe00866000 rw-p 00000000 00:00 0                          [stack]
7ffe0093f000-7ffe00942000 r--p 00000000 00:00 0                          [vvar]
7ffe00942000-7ffe00944000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
process[robot_state_publisher-5]: started with pid [17668]
process[joint_state_publisher-6]: started with pid [17688]
process[joint_state_desired_publisher-7]: started with pid [17694]
[INFO] [1643393282.613947]: Controller Spawner: Waiting for service controller_manager/load_controller
================================================================================REQUIRED process [franka_control-3] has died!
process has died [pid 17644, exit code -6, cmd /home/venkatesh/franka_ws/devel/lib/franka_control/franka_control_node __name:=franka_control __log:=/home/venkatesh/.ros/log/3b3be3d6-8065-11ec-8dcd-3ca82af3b35c/franka_control-3.log].
log file: /home/venkatesh/.ros/log/3b3be3d6-8065-11ec-8dcd-3ca82af3b35c/franka_control-3*.log
Initiating shutdown!
================================================================================
[joint_state_desired_publisher-7] killing on exit
[robot_state_publisher-5] killing on exit
[joint_state_publisher-6] killing on exit
[franka_control-3] killing on exit
[state_controller_spawner-4] killing on exit
[franka_gripper-2] killing on exit
[WARN] [1643393282.896114]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

gollth, 2022-02-03:
Hello @venkateshsripada and thanks for the issue.

First please format ROS logs in code tags in Markdown to be able to read them more easily

gollth, 2022-02-03:
Second: I think this is highly related to https://github.com/frankaemika/franka_ros/issues/106. Your libfranka + franka_ros versions seem quite old. Are you able to upgrade to >= 0.8.0?