frankarobotics/franka_ros #238: Gripper GraspAction command always aborted

hennking98, 2022-03-23:
Hello,
I am trying to grasp something with the gripper using the Simple Action Client. However, I always get the error message: "libfranka gripper: Command aborted!". 

I am initializing the client like this.
```
#include <actionlib/client/simple_action_client.h>
#include <franka_gripper/GraspAction.h>
#include <franka_gripper/MoveAction.h>
#include <franka/gripper_state.h>
#include <franka/gripper.h>

actionlib::SimpleActionClient<franka_gripper::GraspAction> grasp("franka_gripper/grasp", true);
```
In my update function I then send the grasp command like this:
```
    franka_gripper::GraspGoal goal;
    goal.width = 0.03;
    goal.speed = 0.01;
    goal.force = 20;
    goal.epsilon.inner = 0.005;
    goal.epsilon.inner = 0.005;
    grasp.sendGoal(goal);
```
Really rarely the gripper grasps the object, but  most of the time the gripper closes and opens right away. 
Moving the gripper interestingly works without any problems.

I appreciate any help or hint.

marcbone, 2022-03-24:
try to run the homing action once. you can also do this in desk

hennking98, 2022-03-24:
Thanks for the answer. I fixed my issue. I figured out that when I send the move command constantly on every run of my update function, a stop command or a grasp command is not stopping or overwriting the move command. Move commands are still send for a few seconds after executing the grasp or stop command.

closed 2022-03-24 by hennking98

xiaomengfang, 2023-10-22:
@hennking98 Thanks for your feedback. Could you please tell me how to stop the grasp command while the gripper has not moved to the target? For example, I set the gripper width to 0.0 to grasp an object, when it at 0.02, I want to open it to 0.04 immediately.