frankarobotics/franka_ros #240: Incompatible library version (server version: 5, library version: 4)

simonbogh, 2022-03-25:
I just updated one of our robots to 4.2.1 and getting this error with the latest updates when trying to connect ROS to the robot:
`Exception: libfranka: Incompatible library version (server version: 5, library version: 4).`

Versions:
* Robot: 4.2.1
* franka_ros from ROS repo: 0.8.2
* libfranka from ROS repo: 0.8.0

marcbone, 2022-03-25:
We are aware of this. There will be an update for libfranka (0.9.0) on github today. You should be able to link this against the franka_ros repo. There will also be an additional update to franka_ros in the beginning of next week.

simonbogh, 2022-03-25:
Good to hear 👍 

Will you also push a new deb package and when?

marcbone, 2022-03-25:
You mean over the ros package system? Yes, but not today. Also we cannot control when the ros maintainers will distribute it to end-users. This could take a few weeks. However, you can create your own debian package in the meantime:
```bash
git clone --recursive https://github.com/frankaemika/libfranka
cd libfranka
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF ..
cmake --build .
cpack -G DEB
sudo dpkg -i libfranka-0.9.0-amd64.deb
```

Dangmu1996, 2022-03-28:
I am having same problem, even though libfranka is updated to 0.9.0

closed 2022-04-26 by gollth

alison-bartsch, 2022-07-01:
I am getting the same exception error currently. Could you post more detailed instructions as to how to ensure the versions are correct? I am selecting the correct robot firmware option of '4' during the installation.

marcbone, 2022-07-04:
See [https://frankaemika.github.io/docs/compatibility.html](https://frankaemika.github.io/docs/compatibility.html). In the next version of libfranka you will be displayed this link when you get this error.

Kzernobog, 2023-01-17:
Hi, I have built libfranka 0.8.0 in a container and am trying to run the container connected to a robot that runs server version 4 (4.0.2). Technically according to the compatibility document both of these are version 4. However I still get the following error when trying to run the ros control node.

```
Libfranka: Incompatible library version (server version: 4, library version: 5). Please check https://frankaemika.github.io for panda system updates or choose a libfranka version that uses the server version 4 from the...
```

Any ideas as to what I may be missing here?

Ivy-zhouyifan, 2023-07-10:
> Hi, I have built libfranka 0.8.0 in a container and am trying to run the container connected to a robot that runs server version 4 (4.0.2). Technically according to the compatibility document both [...]

I met the same problem and fix the problem by installing it manually via Github.
The reason is `pip install ` only install pre-compiled content. After updating libfranka, you need to compile and package it yourself.

`git clone --recurse-submodules git@github.com:pantor/frankx.git`
`cd frankx`
`mkdir -p build`
`cd build`
`cmake -DBUILD_TYPE=Release .. `
`make`
`make install`
And install frankx via
`pip install .`

Problem Solved! Enjoy!