frankarobotics/franka_ros #241: How to set an external load with SetLoad()?

hennking98, 2022-03-30:
Hello,
I am using the provided cartesian_impedance_example_controller and would like to set a load of a grasped object. I tried the following code, which does not work:
```
#include <franka_msgs/SetLoad.h>
```
```
     ros::ServiceClient client;
     franka_msgs::SetLoad srv;

     client = node_handle.serviceClient<franka_msgs::SetLoad>("set_load");
     srv.request.mass = mass_of_load;
     srv.request.F_x_center_load = gravity_vector; 
     srv.request.load_inertia = inertia_of_load ; 
     client.call(srv);
```
I am new to ROS and highly appreciate any help.
Thanks!

marcbone, 2022-03-31:
> srv.request.F_x_center_load = gravity_vector; 
This is the problem. F_x_center_load describes the translation between the flange and the center of mass of the grasped object. If it is directly between the fingers you can set it to {0,0,0.17}

hennking98, 2022-03-31:
Hi, I figured out that in Gazebo I have to use 
`client = node_handle.serviceClient<franka_msgs::SetLoad>("/set_load");`
while on the real robot it is
`client = node_handle.serviceClient<franka_msgs::SetLoad>("franka_control/set_load");`

However, on the real robot I have the problem, that I cannot overwrite the setLoad command with a new client call. Only a restart of ROS can set the load back to zero. In Gazebo, overwriting works fine.

I would appreciate any help, how I can set back the load to zero on the real robot when I have set it before to a non zero value.

gollth, 2022-04-25:
> I cannot overwrite the setLoad command with a new client call

What exactly do you mean by that? Usually a new `rosservice call /franka_control/set_load ...` should work just fine, both on the real robot and in simulation. Could you give us some more details, like error messages?

Also please note that the `set_load` is a _non-realtime_ command, which cannot be executed successfully when a control loop is running. Refer [to this documentation](https://frankaemika.github.io/docs/libfranka.html#non-realtime-commands) for more information

closed 2022-08-01 by gollth