frankarobotics/franka_ros #244: Can't run move_to_start example

xav12358, 2022-04-05:
Hello,

I try to run move_to_start launch example.

Here is the full log:

roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 
```
... logging to /home/dev/.ros/log/8d87df00-b4ef-11ec-ba51-1bda6a381553/roslaunch-dev-57548.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dev:44965/

SUMMARY
========

PARAMETERS
 * /effort_joint_trajectory_controller/constraints/goal_time: 0.5
 * /effort_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /effort_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /effort_joint_trajectory_controller/gains/panda_joint1/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint1/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint1/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint2/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint2/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint2/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint3/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint3/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint3/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint4/d: 30
 * /effort_joint_trajectory_controller/gains/panda_joint4/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint4/p: 600
 * /effort_joint_trajectory_controller/gains/panda_joint5/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint5/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint5/p: 250
 * /effort_joint_trajectory_controller/gains/panda_joint6/d: 10
 * /effort_joint_trajectory_controller/gains/panda_joint6/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint6/p: 150
 * /effort_joint_trajectory_controller/gains/panda_joint7/d: 5
 * /effort_joint_trajectory_controller/gains/panda_joint7/i: 0
 * /effort_joint_trajectory_controller/gains/panda_joint7/p: 50
 * /effort_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /effort_joint_trajectory_controller/type: effort_controller...
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_limit_warning_threshold: 0.1
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/realtime_config: enforce
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
[... 432 lines cut ...]
log file: /home/dev/.ros/log/8d87df00-b4ef-11ec-ba51-1bda6a381553/move_to_start-9*.log
Initiating shutdown!
================================================================================
[move_to_start-9] killing on exit
[move_group-8] killing on exit
[controller_spawner-6] killing on exit
[joint_state_publisher-5] killing on exit
[robot_state_publisher-4] killing on exit
[state_controller_spawner-3] killing on exit
[franka_control-2] killing on exit
[INFO] [1649170188.439770]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1649170188.440090]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1649170188.440994]: Stopping all controllers...
[INFO] [1649170188.441619]: Stopping all controllers...
[WARN] [1649170188.519227]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[WARN] [1649170188.519233]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

After a few secs, it crashes. I am on ubuntu 20.04 and I use Neotic version of franka_ros.

moshuilanting, 2022-09-07:
hello，I meet the same problem, when I roslaunch the move_to_start example on Ubuntu 18.04 with ros melodic.

Action client not connected: effort_joint_trajectory_controller/follow_joint_trajectory
Action client not connected: franka_gripper/gripper_action

marcbone, 2022-09-07:
Did you install franka_ros from source or over apt?

Maverobot, 2023-02-08:
@moshuilanting @xav12358 Does this issue still exist for you?