frankarobotics/franka_ros #248: Recommended Way of Working in Custom Franka Gripper/Fingers.

suresh-guttikonda, 2022-04-18:
Hi,

I wanted to get the opinion of the developers and the community on using custom franka gripper fingers- in my use case i want to have fingers with additional revolute joint to pick objects instead of default fingers.
I tried to modify sections of code related to `hand/gripper` in hand.xacro, panda_gazebo.xacro and was using `effort_controllers/JointPositionController` for the revolute joints but I am facing the following error. My question is does the franka_ros is additionally overriding the controller code?

```
[INFO] [1650303453.030012, 1.729000]: Controller spawned
[ INFO] [1650303453.949592739, 2.116000000]: Successfully initialized Franka Gripper Controller for joints 'panda_finger_joint1' and 'panda_finger_joint2'
[INFO] [1650303453.951177, 2.117000]: Controller Spawner: Loaded controllers: franka_gripper
[INFO] [1650303454.034601, 2.155000]: Loading controller: panda_left_finger_controller
[INFO] [1650303454.035893, 2.156000]: Started controllers: franka_gripper
[INFO] [1650303455.336057, 2.837000]: Loading controller: panda_right_finger_controller
[INFO] [1650303456.638490, 3.340000]: Controller Spawner: Loaded controllers: joint_state_controller, panda_left_finger_controller, panda_right_finger_controller
[ERROR] [1650303457.151853289, 3.548000000]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
[ERROR] [1650303457.152599, 3.548000]: Failed to start controllers: joint_state_controller, panda_left_finger_controller, panda_right_finger_controller
^C[rviz-10] killing on exit
```

Thanks in advance

marcbone, 2022-04-19:
Do you get this error in gazebo or on the real robot? In gazebo, I inserted some sanity checks in the prepareSwitch method to make sure you dont start malformed controllers. If you modify the urdf to contain an additional joint this will no longer work.

suresh-guttikonda, 2022-04-20:
Hi @marcbone,
Yes i am facing this error in franka_gazebo pkg. I want to modify the default gripper (something similar to as seen in image) so got rid of `franka_gazebo/FrankaGripperSim` controller and related code for prismatic movement in related files.

My custom gripper looks something similar as follows.
``` 
======panda_gazebo.xacro======
    .....
    <link name="${arm_id}_leftfinger_base">
      <visual>
        <geometry>
          <box size="0.01 0.01 0.04"/>
        </geometry>
      </visual>
    </link>
    <joint name="${arm_id}_finger_base_joint1" type="fixed">
      <parent link="${arm_id}_hand"/>
      <child link="${arm_id}_leftfinger_base"/>
      <origin xyz="0 0.05 0.075" rpy="0 0 0"/>
    </joint>
    
    <link name="${arm_id}_leftfinger">
      <visual>
        <geometry>
          <box size="0.01 0.01 0.04"/>
        </geometry>
      </visual>
    </link>
    <joint name="${arm_id}_finger_joint1" type="revolute">
      <parent link="${arm_id}_leftfinger_base"/>
      <child link="${arm_id}_leftfinger"/>
      <origin xyz="0.01 -0.005 0.03" rpy="${pi/10} 0 0"/>
      <limit effort="100" lower="0.0" upper="0.04" velocity="0.2"/>
    </joint>
```
[image]

`If you modify the urdf to contain an additional joint this will no longer work.` => is there any trick to make the above code working ?

marcbone, 2022-04-20:
As we intentionally disallowed this case, you will be on your own implementing what you want. However, I can give you a few pointers for the start:
* To control a joint you have to specify what interface it has in the urdf:
   https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/panda_arm.urdf.xacro#L47
* Then you have to adapt these asserts or remove them: https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L182 However, there are many places where we assume exactly 7 joints. So you will have to go over everything and see what you have to change
* Then the controller verifier could be a problem. It determines whether the controller you selcected is valid. https://github.com/frankaemika/franka_ros/blob/develop/franka_gazebo/src/franka_hw_sim.cpp#L640 You maybe have to change the logic inside the controller verifier to make your use-case work.

suresh-guttikonda, 2022-04-20:
@marcbone thank you for the helping with the hints.

gollth, 2022-04-21:
@suresh-guttikonda consider closing this ticket when it solved your problem

closed 2022-04-22 by suresh-guttikonda