frankarobotics/franka_ros #251: Incompatible library version (server version: 4, library version: 5)

rhaschke, 2022-05-02:
Trying to run latest franka_ros (version 0.9.0) on an older robot system version (4.2.0), yields the following error message from libfranka:
```
[ERROR] ros.franka_hw: FrankaHW: Failed to initialize libfranka robot. libfranka: Incompatible library version (server version: 4, library version: 5). Please check https://frankaemika.github.io for Panda system updates or use a different version of libfranka.
[ERROR] ros.franka_control: franka_control_node: Failed to initialize FrankaHW class. Shutting down!
```
Updating to robot version 4.2.1 solved that issue. However, the error message, mentioning server version 4 vs. library version 5 was rather confusing. I would have expected mentioning 4.2.1...

marcbone, 2022-05-02:
Yeah it is a bit confusing. There is a difference between "server" version and "system" version. The server version is actually a API compatibility indicator. We have a table [here](https://frankaemika.github.io/docs/compatibility.html) to find out the correct version for your robot. In the future, the error message that you got will also contain this link.

rhaschke, 2022-05-02:
Thanks. Having this link in the error message will be helpful.

closed 2022-05-02 by rhaschke