frankarobotics/franka_ros #252: libfranka: unable to set realtime scheduling: Operation not permitted

rhaschke, 2022-05-02:
Trying to activate any controller, I get the following exception:
`libfranka: unable to set realtime scheduling: Operation not permitted`

Of course, realtime limits are correctly [configured in /etc/security/limits.conf](https://frankaemika.github.io/docs/installation_linux.html#allow-a-user-to-set-real-time-permissions-for-its-processes) (and yet, I'm member of group `nirobots`):
```
@nirobots        -       rtprio          99
@nirobots        -       priority        99
@nirobots        -       memlock         102400
```
Have these required settings changed (because it worked with version 0.8.x)?

marcbone, 2022-05-02:
There shouldn't be any change regarding these settings (I have the same on my machine). Are you running a rt-kernel?
If you think that it is really due to a change in libfranka and have another robot that you didnt update yet you can try to install libfranka 0.8.0 again:

```bash
sudo apt remove ros-noetic-libfranka
git clone --recursive https://github.com/frankaemika/libfranka --branch 0.8.0
cd libfranka
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF ..
cmake --build .
cpack -G DEB
sudo dpkg -i libfranka-0.8.0-amd64.deb
```
If you do that you need to install franka_ros from sources, but I guess you do that anyway.

rhaschke, 2022-05-02:
> Are you running a rt-kernel?

Argh. That's it! I didn't boot the RT kernel...

closed 2022-05-02 by rhaschke