frankarobotics/franka_ros #255: Franka URDF With Course Collision Model

fishbotics, 2022-05-09:
Hi all, 

I run the robot in simulation (usually Pybullet) and check collisions with the collision meshes. I've seen that there has been discussion about using inflated self-collision meshes to check self-collisions. Does anyone have a URDF with the improved self collision model? I've seen it in several custom `xacro` files but I'm not sure how to parse those to produce a URDF with all the special custom tags.

Thanks a lot

marcbone, 2022-05-10:
That is a tricky task. In theory you could do `xacro panda_arm.urdf.xacro hand:=true`. However, the resulting URDF would contain both collision models. If you would load this model in pybullet, it would use this model for physical interaction and that is probably not what you want.
[image]

 You want to use the "inflated self collision model" only for path planning, which you have to do outside of bullet anyways (Also you have to disable collisions between a few links).

fishbotics, 2022-05-10:
Thanks @marcbone , I typically do planning with raw OMPL using Pybullet for collision checking. What would be ideal would be to use one urdf for obstacle collision and one urdf for self collision (maybe in parallel simulations to avoid weird physics with two robots).

marcbone, 2022-05-10:
I never worked with OMPL directly. I guess you can load a URDF into it for collision checking? There you would put the URDF that you generated with `xacro panda_arm.urdf.xacro hand:=true`. Then you would have to disable collision between links according to [this config from moveit](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/config/arm.xacro)

fishbotics, 2022-05-10:
It works through callbacks. You set up a collision checking function and pass it into the planner which then uses that to verify states. This is why I also use Pybullet--it's a fairly lightweight collision checker that can read in a URDF. I could see some other lightweight checker being more appropriate but I haven't found one that's as easy to use from Python. Hence why I'm asking about a urdf with the inflated collision shapes.

marcbone, 2022-05-10:
Ah ok. Then I guess you can use the inflated URDF and use setCollisionFilterPair function in pybullet to disable the unwanted collisions between links

gollth, 2022-07-07:
Hi @fishbotics 

you can use

```bash
xacro $(rospack find franka_description)/robot/panda_arm.urdf.xacro hand:=true gazebo:=true
```

This will omit the `*_sc` capsules (although it will provide empty links to keep the URDF consistent) from the URDF. If you find it confusing why we named this arg `gazebo` please refer to the [docs on develop](https://github.com/frankaemika/docs/blob/develop/source/franka_ros.rst#franka_description)

Let me know if this helps

closed 2022-08-01 by gollth