frankarobotics/franka_ros #256: How to use the gripper

arnauddomes, 2022-05-16:
Hello,

I want to use the gripper in python. I follow the franka ros tutorial and I get that command to use the gripper:

`rostopic pub --once /franka_gripper/grasp/goal              franka_gripper/GraspActionGoal              "goal: { width: 0.022, epsilon:{ inner: 0.005, outer: 0.005 }, speed: 0.1, force: 5.0}"`

Is there a moveit message to command the gripper ?

I doesn't find any python example for that.

marcbone, 2022-05-18:
The grasping is a ros action. You have to look at the ros action tutorials. For your convenience, I took the example from there and adapted it for the grasping message:

```python
#! /usr/bin/env python3
import franka_gripper.msg
import rospy
import sys

# Brings in the SimpleActionClient
import actionlib

# Brings in the messages used by the grasp action, including the
# goal message and the result message.

def grasp_client():
    # Creates the SimpleActionClient, passing the type of the action
    # (GraspAction) to the constructor.
    client = actionlib.SimpleActionClient('/franka_gripper/grasp', franka_gripper.msg.GraspAction)

    # Waits until the action server has started up and started
    # listening for goals.
    client.wait_for_server()

    # Creates a goal to send to the action server.
    goal = franka_gripper.msg.GraspGoal()
    goal.width = 0.022
    goal.epsilon.inner = 0.005
    goal.epsilon.outer = 0.005
    goal.speed = 0.1
    goal.force = 5

    # Sends the goal to the action server.
    client.send_goal(goal)

    # Waits for the server to finish performing the action.
    client.wait_for_result()

    # Prints out the result of executing the action
    return client.get_result()  # A GraspResult

if __name__ == '__main__':
    try:
        # Initializes a rospy node so that the SimpleActionClient can
        # publish and subscribe over ROS.
        rospy.init_node('grasp_client_py')
        result = grasp_client()
        print("Success: ",result.success)
        print("Error message: ", result.error)
    except rospy.ROSInterruptException:
        print("program interrupted before completion", file=sys.stderr)

```

closed 2022-05-18 by marcbone

InsightofSPb, 2023-03-06:
Hello,
I am working on project, connected with this gripper. Thanks for the code above, it helped to understand many thing. Unfortunately, when I run it, I receive an error:
```
ModuleNotFoundError: No module named 'franka_gripper'
```
However, `import franka_gripper.msg` is added like in the code, discussed in this question.....

marcbone, 2023-03-07:
you need to source ros and if you compile franka_ros yourself you also need to source the setup.bash in your devel folder

InsightofSPb, 2023-03-07:
It worked. Thank you so much! 
P.S. There were some problems with my system, but you've set me on the right path

NevermoreFilo, 2024-04-09:
> The grasping is a ros action. You have to look at the ros action tutorials. For your convenience, I took the example from there and adapted it for the grasping message:
>
> ```python
> #! /usr/bin/env [...]

Does this code expect you to also execute a server starting call somewhere?