frankarobotics/franka_ros #258: panda.launch/ name 'xacro' is not defined ??

jasser-enbaya, 2022-05-23:
Hello,
I work with Ubuntu 18.04 (Ros melodic)
I installed Libfranka and franka-ros from this site : https://frankaemika.github.io/docs/index.html  
and I didn't face any problem or error. After that, I source my workspace and I want to run the command in the franca_gazebo tutorial, which is : 
roslaunch franka_gazebo panda.launch x:=-0.5 \
    world:=$(rospack find franka_gazebo)/world/stone.sdf \
    controller:=cartesian_impedance_example_controller \
    rviz:=true

But the launch file don't work, and the output was like that: 

*****************************************************************************************************************************************
jesser@jesser-meca:~/catkin_ws$ roslaunch franka_gazebo panda.launch x:=-0.5     world:=$(rospack find franka_gazebo)/world/stone.sdf     controller:=cartesian_impedance_example_controller     rviz:=true
... logging to /home/jesser/.ros/log/0c46e10e-da90-11ec-b709-3ca067ee1911/roslaunch-jesser-meca-27581.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

name 'xacro' is not defined 
when evaluating expression 'xacro.load_yaml('inertial.yaml')'
when instantiating macro: link_with_sc (/home/jesser/catkin_ws/src/franka_ros/franka_description/robots/utils.xacro)
instantiated from: panda_arm (/home/jesser/catkin_ws/src/franka_ros/franka_description/robots/panda_arm.xacro)
in file: /home/jesser/catkin_ws/src/franka_ros/franka_description/robots/panda_arm.urdf.xacro
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['xacro', '/home/jesser/catkin_ws/src/franka_ros/franka_description/robots/panda_arm.urdf.xacro', 'gazebo:=true', 'hand:=true', 'arm_id:=panda', 'xyz:=-0.5 0 0', 'rpy:=0 0 0']] returned with code [2]. 

Param xml is <param command="xacro $(find franka_description)/robots/panda_arm.urdf.xacro                   gazebo:=true                   hand:=$(arg use_gripper)                   arm_id:=$(arg arm_id)                   xyz:='$(arg x) $(arg y) $(arg z)'                   rpy:='$(arg roll) $(arg pitch) $(arg yaw)'" name="robot_description">
  </param>
The traceback for the exception was written to the log file
*******************************************************************************************************************************************************

I didn't change anything in files, so please can someone explain this error from me, i searched in interned and i didn't find any prob similar to mine, THX in advance.

closed 2022-06-02 by jasser-enbaya

HongminWu, 2022-07-20:
how to solve this problem?

Ekanshh, 2022-08-28:
@HongminWu In `franka_description`, check your branch. For me, it was on `develop` and not on `melodic-develop`. So I did checkout to `melodic-devlop`. Then I build the workspace again and source it. It worked fine for me afterwards. :)

yuqiang6149, 2022-12-07:
I also met this problem when i was running the panda_moveit_config package. And it seems to be a problem about the xacro version. So i clone the xacro package source into my workspace and remake.

Jasser-123, 2024-02-19:
Thank you all for your responses. I apologize for not being active on this matter.
If I recall correctly, the issue stemmed from the version of the xacro after I executed the `sudo apt-get upgrade` command. Consequently, I uninstalled it and reinstalled the default version, which resolved all the errors.

So check if you upgrade anything, maybe it's not the xacro it could be **franka_description**.