frankarobotics/franka_ros #260: Libfranka library and server incompatibility

Obscuriosity, 2022-05-25:
Trying to launch the ROS visualisation or echo_robot_state generates the following error:

libfranka: Incompatible library version (server version: 5, library version: 4). Please check https://frankaemika.github.io for Panda system updates or use a different version of libfranka.

I am running Ubuntu 16.04 and ROS kinetic.
libfranka is 0.8.0-1xenial
System version is 4.2.1
Do I need to roll back my server/system version?

Thanks

marcbone, 2022-05-27:
Officially, we do not support ROS kinetic anymore. I think we made a few changes in franka_ros could cause you some troubles. Apart from that, your problem is that you need to upgrade you libfranka to version 0.9.

closed 2022-05-27 by marcbone

Obscuriosity, 2022-05-30:
Hello marcbone
You may wish to update this page: https://frankaemika.github.io/docs/installation_linux.html.
It says you provide official support for: Ubuntu 16.04 LTS Xenial Xerus and ROS Kinetic Kame.
Thanks for the reply, I will update to 20.04 and ROS Noetic.

marcbone, 2022-05-30:
Oh yeah this is true. Thank you. We should update that.