frankarobotics/franka_ros #265: Stuck in "updating joint firmware" when starting robot

ljhnick, 2022-06-30:
After I turn the power on, the light on the arm will not be on.
And if I connect the PC with the control via LAN and access the 172.16.0.2 for the DESK interface, it is stuck and shows "updating joint firmware" and I cannot do anything else. Does anybody have this issue before? I am wondering how can I solve this issue?
[image]

ljhnick, 2022-07-01:
Actually I solved it. This is a very weird situation. There is a "functional earth" on the base of the robotic arm and if you clamp that part to fix the arm, it will stuck. 
So if you have the same issue, just check the earth part.
Hopefully it is helpful.

closed 2023-02-07 by Maverobot

sundtek, 2026-11-04:
Hi, I might be experiencing a similar issue with my Franka Panda, but I'm not quite sure what you mean by "clamping that part to fix the arm."

Could you please elaborate on what exactly caused the problem and how you fixed it? Are you referring to the functional earth (grounding) connection at the base of the robot?

Thanks in advance for your help!

ljhnick, 2026-11-04:
> 3\. Hi, I might be experiencing a similar issue with my Franka Panda, but I'm not quite sure what you mean by "clamping that part to fix the arm."
>     Could you please elaborate on what exactly caused the problem and how you fixed it? Are you referring to the functional earth (grounding) connection at the base of the robot?
>     Thanks in advance for your help!

it is too long time ago and i cannot recall what exactly happened. What i was trying to do is probably that i need to fix the robot on top of a table and to do that i need a C-clamp to clamp the base of the robot tightly to the table top. I am not sure if that is the issue but that is all i can recall.