frankarobotics/franka_ros #266: Get user button state

xav12358, 2022-07-07:
Hello,

I want to catch user button interaction on the last joint. I know that we can collect the state of the robot_mode using FrankaState message.

Is there an other message to get the state of the button on the last joint?

gollth, 2022-07-08:
Hi @xav12358 

The robot has many buttons at the pilot. Which one do you mean exactly?

xav12358, 2022-07-08:
I mean these buttons.
[image]

gollth, 2022-07-08:
The state of the pilot buttons cannot be accessed via the FCI. If you need them I suggest to use the App interface of our robot. Refer: https://github.com/frankaemika/the_reference_platform/tree/main/2_RIDE

Good luck!

closed 2022-07-08 by gollth