frankarobotics/franka_ros #268: ros_control error during Catkin_make of franka_ros package on Arch Linux real-time kernel

ruturajsambhusvt, 2022-07-12:
Hello, I have been trying to catkin_make the franka_ros package and the following error is obtained (also attached CMakeOutput.log)-

CMake Error at /home/trec/catkin_ws/devel/share/controller_interface/cmake/controller_interfaceConfig.cmake:113 (message):
Project 'controller_interface' specifies
'/home/trec/catkin_ws/src/ros_control/controller_interface/include' as an
include dir, which is not found. It does neither exist as an absolute
directory nor in
'/home/trec/catkin_ws/src/ros_control/controller_interface//home/trec/catkin_ws/src/ros_control/controller_interface/include'.
Check the issue tracker
'https://github.com/ros-controls/ros_control/issues' and consider creating
a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
/opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
franka_ros/franka_hw/CMakeLists.txt:7 (find_package)
[CMakeOutput.log](https://github.com/ros-controls/ros_control/files/9088104/CMakeOutput.log)

-- Configuring incomplete, errors occurred!
See also "/home/trec/catkin_ws/build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:2102: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

I have installed the ros_control package using yay -S ros-noetic-ros-control and there were no errors during the installation. It would be great if you can advise on resolving this. I had raised issue with ros_control and they say that it is not a problem with ros_control, but rather with franka_ros.
[CMakeOutput.log](https://github.com/frankaemika/franka_ros/files/9093528/CMakeOutput.log)

gollth, 2022-08-01:
Hi @ruturajsambhusvt 

thanks for your interest in `franka_ros`. Could you please format your console output to use code tags?

I'm no expert in Arch Linux but from your error it looks like that you have a local copy of ros_control in your workspace (at `/home/trec/catkin_ws/src/ros_control` which is compiled during `catkin_make`. Have you tried removing it from your workspace or `CATKIN_IGNORE` it?

ruturajsambhusvt, 2022-08-16:
Hello, thank you for your response. I somehow got it working. Probably there were dependency issues and version incompatibilities that had to be resolved and things started building.

closed 2022-08-16 by ruturajsambhusvt