frankarobotics/franka_ros #272: Could not switch controllers

BolunDai0216, 2022-07-22:
Hi,

I was trying to start creating my own custom controller. So the first thing I did was to try and create a really simple .yaml file with the standard controller types:

```yaml
panda:
  joint1_position_controller:
    type: effort_controllers/JointPositionController
    joint: panda_joint1
    pid: { p: 100.0, i: 0.01, d: 10.0 }
```

However, I got the following error:

```console
[ERROR] [1658503332.265323172]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
[ERROR] [1658503332.265997, 0.000000]: Failed to start controllers: franka_state_controller, panda/joint1_position_controller
```

Is it possible to help me see what the issue is here? I did not find much information online. I thought since `joint1` has

```xml
  <transmission name="panda_joint1_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="panda_joint1">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="panda_joint1_motor">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </actuator>
  </transmission>
```

the hardware interface and controller combination should work.

Thanks in advance!

BolunDai0216, 2022-07-25:
I noticed that I did not perform the step:

```cpp
effort_joint_handles_[i] = effort_joint_interface_->getHandle(joint_names[i])
```

after adding it, the error goes away. Can anyone help me understand why this is the case?

Thanks!

gollth, 2022-08-01:
Hi there,

it's a bit hard to debug your code, when you only provide one liners but I will try anyways:

> I did not find much information online

This should be a good place to start: http://wiki.ros.org/ros_control#Overview

> I was trying to start creating my own custom controller. So the first thing I did was to try and create a really simple .yaml file with the standard controller types.

I don't fully understand what you are trying to do. Are you writing your own controller or are you using the already existing [effort_controllers/JointPositionController](http://docs.ros.org/en/jade/api/effort_controllers/html/classeffort__controllers_1_1JointPositionController.html)?

> The hardware interface combination for the requested controllers is unfeasible.

This error indicates that one (or multiple) controller(s) that you are trying to start what to claim commanding interface of the robot which exclude each other. For example you cannot claim a commanding position & velocity interface at the same time, since our robot cannot follow both.

> I thought since joint1 has ... the hardware interface and controller combination should work.

This is correct

> Can anyone help me understand why this is the case? 

Again hard to say without the whole code. Can it simply be that you try to launch the same controller multiple times?

gollth, 2022-08-01:
In any case this seems more like a issue related to your code not to `franka_ros` itself. Should this be the case, though, feel free to reopen this ticket

closed 2022-08-01 by gollth