frankarobotics/franka_ros #273: Cartesian velocity control with dual arm setup

hmccarty, 2022-07-27:
We have two Panda systems in a workspace and would like to control both using cartesian velocity control.

What's the best way of accomplishing this? If we don't intend on controlling both at the same time, can we get away using a single controller which implements the `FrankaVelocityCartesianInterface`? We tried doing this, but unfortunately it showed that the interface wasn't available.

We also saw that the documentation mentioned using both the effort and cartesian velocity interface when running robots that implement `FrankaCombinableHW`, see table below, but other areas of the documentation contradict this.

[image]

Can anyone offer any help or advice?

rhaschke, 2022-07-27:
Controlling two robots independently in ROS is usually accomplished by introducing corresponding ROS namespaces,
i.e. running each robot in its own namespace.
If you want both arms to be combined in a single `robot_description` instead, have a look at the example [`franka_combined_control.launch`](https://github.com/frankaemika/franka_ros/blob/54ad920ed6a69e7552bfa993d0b063758f9254cb/franka_control/launch/franka_combined_control.launch)

closed 2022-07-27 by hmccarty

mitsui29, 2024-02-28:
Hello,
Great talks! I wanna start with dual arm on gazebo and real panda for a cooperative manipulation. To finish this task i will use some methods based on the computed Joints Torques(in c++). First i wanna try this on the gazebo and then in the real panda. But i am quite new for this stuff. I find, in franka_ros there is no gazebo part for dual arm. Can anyone helps me to find a way to handle this problem?  Thanks a lot!