frankarobotics/franka_ros #280: Access state of pilot buttons?

rhaschke, 2022-08-24:
Is it possible to access the state (and lighting) of the pilot buttons via ROS/FCI?

Maverobot, 2023-02-07:
Unfortunately it is currently not possible.

rhaschke, 2023-02-07:
What is missing? Is that state available in libfranka and only the corresponding exposure to ROS is missing or is the problem more fundamental and FCI doesn't provide that information yet?

Maverobot, 2023-02-07:
The information of the pilot/leds is not exposed via FCI at all. We are aware of this feature request but it is currently not prioritized.

rhaschke, 2023-02-07:
I see. Thanks for the clarification.

jellehierck, 2025-02-14:
There is a workaround available to access the buttons by connecting to the web API of Desk, like in this library: <https://github.com/JeanElsner/panda-py>. `panda-py` provides Python bindings for many Desk functionality, including reading the pilot buttons.

The `franka_buttons` repository (<https://github.com/franzesegiovanni/franka_buttons>) took a subset of `panda-py`'s bindings to just obtain the buttons and publish information to ROS 1.

I ported `franka_buttons` to ROS 2 (<https://github.com/jellehierck/franka_buttons_ros2>) and refined the idea some more.

The idea is to connect to the Desk headlessly, i.e. log in to Desk and opening a websocket to obtain button events in real-time. This can capture button events at any time, even alongside FCI. There are some smal caveats, e.g. the arrow buttons (up, down, left and right) are not available when FCI is enabled, and you need to relinquish Desk control to the node (meaning you cannot use the Desk dashboard in the browser while the node is running).

I have not personally tested the ROS 1 version but I am actively developing my port `franka_buttons_ros2`. Maybe it is useful for someone who comes across this, any feedback is welcome in the [issues](https://github.com/jellehierck/franka_buttons_ros2/issues) :)