frankarobotics/franka_ros #289: What is the purpose of filtering the joint velocity measurements?

BolunDai0216, 2022-10-06:
Hi,

I noticed that in `joint_impedance_example_controller`, instead of directly using the measured joint velocities `robot_state.dq[i]` it uses a filtered version instead:

```cpp
for (size_t i = 0; i < 7; i++) {
  dq_filtered_[i] = (1 - alpha) * dq_filtered_[i] + alpha * robot_state.dq[i];
}
```

can anyone help me understand what is the purpose of doing this? 

Thanks in advance!

marcbone, 2022-10-07:
the "measured velocity" is actually the time derivative of the measured position. As the measured position is subject to some noise, the velocity is also a bit noisy. Therefore, to avoid oscillation the example uses the filter to avoid jerky motions. However, feel free to use the measured velocity directly and see if it still works

BolunDai0216, 2022-10-07:
> the "measured velocity" is actually the time derivative of the measured position. As the measured position is subject to some noise, the velocity is also a bit noisy. Therefore, to avoid oscillation [...]

Thanks for the response!

I just tested using `robot_state.dq` directly by setting `alpha = 1.0`. It worked fine without any jerky motions.

closed 2022-10-07 by BolunDai0216