frankarobotics/franka_ros #295: Too much Noise in the simulated joint torque signals

ZhaotingLi, 2022-11-03:
Hi,

I was doing some torque-related project using Franka. I found that the torque data in the simulation is rather noisy than the torque data from the real robot. As shown in the figure below, the robot arm moves randomly and there is an external force acting on it since around 20 second. However, the noise level of the torque is rather high and seems unreasonable. Is there anyone know the reason? Thanks in advance! 

[image]

Here is another figure where the robot is static. 
[image]

andrejpan, 2022-11-06:
Hey, 
could you tell us which simulation you are using? Maybe also provide a repository?

Maverobot, 2023-02-07:
The noise is from the simulator, which is likely unrelated to franka_ros.

Closed due to inactivity.

closed 2023-02-07 by Maverobot