frankarobotics/franka_ros #297: Strange external force and wrench from real robot

Lejou, 2022-11-08:
Env: ROS Noetic

I start up the model_example_controller.launch in the terminal with a real robot and I obtained the weird F_ext with a large value on z. I think it has some relationship with the unmounted gripper, since when I run the simulation with gripper in gazebo, the F_ext is correct.

 Anyway, how to figure it? I really need a help.

[image]

marcbone, 2022-11-08:
The robot still thinks you have a gripper attached. You will also feel that the robot will try to move up when you hand guide the robot. You have to go into settings in DESK (the webinterface) go to end-effector and set "None" as end-effector.

Lejou, 2022-11-09:
Thank you for your help.

closed 2022-11-09 by Lejou