frankarobotics/franka_ros #307: What difference between the external controller and the internal controller? And how to use external controller in Franka_ros?

Lejou, 2022-12-04:
Hello.

I do go through the doc about franka in https://frankaemika.github.io/docs/libfranka.html. And I found that Franka provides three controllers, including: external controller, internal joint impedance (which could receive the joint torque commands, I guess ), and internal Cartesian impedance . Currently, I use internal joint impedance so that I can send joint torque commands to the robot. But I have no idea about the other two controllers. Could someone give an explanation or some information? And, what the difference between the external controller and internal controller? In my view, the external controller could receive joint torque commands with gravity, while the internal impedance controller could compensate the gravity automatically. So, the gravity seems to be the difference, is that correct?

Moreover, in franka control/config/franka_control_node.yaml, there is only an option for internal controllers. what if I want to use the external controller? Does we have the related interface in this package?

Maverobot, 2023-02-07:
> Could someone give an explanation or some information? And, what the difference between the external controller and internal controller? 

Please check examples in franka_ros and libfranka:

- https://github.com/frankaemika/franka_ros/tree/develop/franka_example_controllers/src
- https://github.com/frankaemika/libfranka/tree/master/examples

> In my view, the external controller could receive joint torque commands with gravity, while the internal impedance controller could compensate the gravity automatically. So, the gravity seems to be the difference, is that correct?

When using internal controllers, the user just have to feed the robot with desired JointPosition, JointVelocity, CartesianPose or CartesianVelocity every millisecond. The internal controllers will take care of executing the motion including compensating the gravity.

When using external controllers, the user must write the controller by themselves, i.e. by writing a callback mapping robot state to tau_J_d. However, the tau_J_d, which is computed by the user defined control callback, should not contain gravity compensation since the gravity term will added automatically in the robot. 

> Moreover, in franka control/config/franka_control_node.yaml, there is only an option for internal controllers. what if I want to use the external controller? Does we have the related interface in this package?

No addition interface is needed for external controller. This `internal_controller` variable only specifies which internal controller to use if the user does not define an external controller.

Maverobot, 2023-02-07:
I will close it for now. Feel free to reopen it if there are still confusions after going through the documentation and code examples.

closed 2023-02-07 by Maverobot