frankarobotics/franka_ros #312: libfranka: Incompatible library version (server version: 4, library version: 5)

xzhuzhu, 2023-01-01:
The libfranka is set 0.8.0, why the error is library version: 5. 
I have used sudo apt remove "*libfranka*" and repeat building libfranka many times.
The ubuntu is 20.04.

andrejpan, 2023-01-03:
Did you recently check the compatibility list?
https://frankaemika.github.io/docs/compatibility.html
I am assuming that the Robot system version on MC is too new (>= 4.2.1) and is not compatible with libfranka 0.8.0.

I would also advise that you start using libfranka >= 0.10.0.

Maverobot, 2023-02-07:
I will close this now. Free free to reopen it if the issue still exists after following the [compatibility list](https://frankaemika.github.io/docs/compatibility.html).

closed 2023-02-07 by Maverobot