frankarobotics/franka_ros #317: error recovery takes long time

HumbertoE, 2023-01-18:
### Context
System: Franka Disruption 3
System version: 5.2.1
libfranka version: 0.10.0
franka_ros version: 0.10.1
controller: position_joint_trajectory_controller

### Problem
When an error happens (in particular cartesian reflexes as this is what we tested), the system takes a long and varying time to confirm it. This comes in two parts:

1.  There is a significant time difference between the manipulator stopping during a trajectory (we carefully blocked the movement of the robot with our hands) and the cartesian reflex being reported by libfranka. Experimentally we found this can be up to 1 or 2 seconds.
2. Once the error is reported, we identify it through the robot mode message attribute published in /franka_state_controller/franka_states. When the robot mode is not "normal" (IDLE or MOVE), we check it and confirm it. A simplified version of the script to generate the ROS node we use for this can be find below.

The time difference between the error recovery request to be sent and the error recovery varies a lot.  In an experiment where we confirmed 11 errors, we found the following time differences in seconds:

1.0152
0.9475
15.9851
2.5994
0.3952
0.8022
1.2532
0.5220
2.1730
0.6632
3.0928

It can be seen that the confirmation is sometimes very quick, other times a bit longer and rarely it takes a very long time.

Part 2 of the problem is more critical for us. 

### Steps to reproduce part 2

1. Start all necessary nodes to run the robot including the franka_control node
4. Start the manipulator_error_manager node with the script below
5. Move the robot and produce a cartesian reflex by carefully blocking the robot's movement
6. Compare the timestamps of the logs printed by libfranka libfranka: Move command aborted: motion aborted by reflex! ["cartesian_reflex"] and `Recovered from error`
7. Repeat steps 5 and 6 multiple times

### Steps to reproduce part 1
1. Start all necessary nodes to run the robot including the franka_control node
4. Start the manipulator_error_manager node with the script below
5. Move the robot and produce a cartesian reflex by carefully blocking the robot's movement
6. Observe how much time it takes for the robot light to turn red
7. Repeat steps 5 and 6 multiple times

### Script of the node to confirm errors
```
#!/usr/bin/python3

import rospy
import traceback
import actionlib
import franka_msgs.msg as franka_msg

class ManipulatorErrorManager:
    '''
    class containing code the get the status of the panda manipulator and chnage this state (mainly confirming errors)
    '''
    def __init__(self):
        # Robot modes and their meaning:
        self.robot_modes_meaning = {'0': 'OTHER',
                                    '1': 'IDLE',
                                    '2': 'MOVE',
                                    '3': 'GUIDING',
                                    '4': 'REFLEX',
                                    '5': 'USER_STOPPED',
                                    '6': 'AUTOMATIC_ERROR_RECOVERY'}
        
        self.confirmation_cooldown_time = 0.2 # [s] cooldown time after an error confirmation attempt

        # Init attributes        
        self.confirmation_time_stamp = rospy.Time.now() # time stamp the of the last confirmation attempt

        # Init subscriber to franka state 
        rospy.Subscriber('/franka_state_controller/franka_states', franka_msg.FrankaState, self.__callback_status_information, queue_size=1)

        # Initialize action client for error recovery
        self.error_recovery_client = actionlib.SimpleActionClient('/franka_control/error_recovery', franka_msg.ErrorRecoveryAction)
        self.error_recovery_client.wait_for_server()
    
    def __callback_status_information(self, msg: franka_msg.FrankaState) -> None:
        ''' Callback method of the subscriber to the franka status topic
        - @param msg: the message that was send by the franka state interface 
        Documentation about the status message: https://frankaemika.github.io/libfranka/structfranka_1_1RobotState.html
        '''
        duration_since_last_confirmation = (rospy.Time.now() - self.confirmation_time_stamp).to_sec()

        # If robot has an error and if not in cooldown time, confirm the error
        if not msg.robot_mode in [1, 2] and duration_since_last_confirmation > self.confirmation_cooldown_time:
            robot_mode_description = self.robot_modes_meaning[str(msg.robot_mode)]
            rospy.logwarn(f"Manipulator error detected. Robot mode: {robot_mode_description}")

            # Define action message
            goal_msg = franka_msg.ErrorRecoveryActionGoal()

            # Trigger action server -> confirm error
            self.error_recovery_client.send_goal_and_wait(goal_msg)

            # Reset confirmation time stamp
            self.confirmation_time_stamp = rospy.Time.now()
            
            rospy.loginfo("Manipulator error confirmed")

def main():
    try:
        rospy.loginfo('Starting Manipulator Error Manager Node')

        rospy.init_node('manipulator_error_manager')

        # Create error manager object
        error_manager = ManipulatorErrorManager()

        rospy.spin()

    except (rospy.ROSInterruptException, KeyboardInterrupt):
        return

    except:
        rospy.logfatal('uncaught error in manipulator_error_manager.py')
        rospy.logfatal(traceback.format_exc())
        return

if __name__ == '__main__':
    main()
```

HumbertoE, 2023-02-07:
> I did not know that we had a system with name of "Disruption". :)

From our understanding, it is a FP3 with FCI

> There is no "step 6". Maybe it was meant to be something else?

I corrected it. I changed the steps while writing them and forgot to update this last step.

> Can you please further check how much time each of these steps are taking? It would be very helpful

It is not straight-forward for us to get the points 2 and 3 that you mention, but from the logs of the experiment I did before, i can get the times of the other 3 points, that is: 

1. Cartesian Reflex, triggered by human interaction, is logged by libfranka with `Move command aborted....`
2. My script's subscriber callback is triggered and the error recovery action is called.
3. `Recovered from error` is logged, I think by the franka_control_node

The duration or time intervals between 1 and 3 are the times mentioned before:

> In an experiment where we confirmed 11 errors, we found the following time differences in seconds:
> 
> 1.0152 0.9475 15.9851 2.5994 0.3952 0.8022 1.2532 0.5220 2.1730 0.6632 3.0928

This table summarizes the durations:

Timestamp of 1 [s] | Timestamp of 2 [s] | Timestamp of 3 [s] | Duration from 1 to 2 [s] | % | Duration from 2 to 3 [s] | % |  Duration from 1 to 3 [s]
-- | -- | -- | -- | -- | --| -- | --
1674054233.4735 | 1674054233.4757 | 1674054234.4887 | 0.0022 | 0.21% | 1.0130 | 99.79% | 1.0152
1674054239.9155 | 1674054239.9174 | 1674054240.8630 | 0.0019 | 0.20% | 0.9456 | 99.80% | 0.9475
1674054244.6415 | 1674054244.6441 | 1674054260.6266 | 0.0026 | 0.02% | 15.9825 | 99.98% | 15.9851
1674054658.3262 | 1674054658.3282 | 1674054660.9256 | 0.0020 | 0.08% | 2.5974 | 99.92% | 2.5994
1674054665.1468 | 1674054665.1502 | 1674054665.5420 | 0.0034 | 0.86% | 0.3918 | 99.14% | 0.3952
1674054670.5906 | 1674054670.5922 | 1674054671.3929 | 0.0016 | 0.20% | 0.8007 | 99.80% | 0.8022
1674054677.2169 | 1674054677.2263 | 1674054678.4701 | 0.0094 | 0.75% | 1.2438 | 99.25% | 1.2532
1674054681.0607 | 1674054681.0636 | 1674054681.5827 | 0.0029 | 0.56% | 0.5191 | 99.44% | 0.5220
1674054941.4138 | 1674054941.4158 | 1674054943.5868 | 0.0020 | 0.09% | 2.1710 | 99.91% | 2.1730
1674054951.7425 | 1674054951.7450 | 1674054952.4057 | 0.0025 | 0.37% | 0.6607 | 99.63% | 0.6632
1674054955.5605 | 1674054955.5621 | 1674054958.6533 | 0.0015 | 0.05% | 3.0912 | 99.95% | 3.0928

With this information, I would say the problem is in the operations done by the actions server

Maverobot, 2023-02-08:
I looked deeper into the issue and found out that the most of the time cost came from the [mutex lock](https://github.com/frankaemika/franka_ros/blob/develop/franka_control/src/franka_control_node.cpp#L52) in the error recovery action callback. It can be conveniently verified by adding a `ROS_INFO` before and after that line.

#### Possible Reason

The mutex is locked when the `FrankaHW.control` is running. After a reflex is triggered, the robot slows down and tries to stop. Only after the "stopping process" finishes, `FrankaHW.control` will be aborted via an exception and release the mutex. Only at this point, the [mutex](https://github.com/frankaemika/franka_ros/blob/develop/franka_control/src/franka_control_node.cpp#L52) can be required and `robot.automaticErrorRecovery();` is executed.

HumbertoE, 2023-02-17:
> Then it is called "Franka Research 3" instead of "Franka Disruption 3" :D

In franka world the name of the system is Franka Disruption 3. Not sure what the difference is. The robot is black, but we can use the FCI.

> In short, the "stopping" of the robot takes too long, which blocks the error recovery action.

Thank you for looking into it. Were you able to verify this was the reason? And what can be done to improve the time it takes?

Maverobot, 2023-02-17:
> In franka world the name of the system is Franka Disruption 3

That is indeed a weird name. @Felix-Kuperjans @fwalch Any idea whether that is intentional?

> Were you able to verify this was the reason?

Yes

> And what can be done to improve the time it takes?

As a user, I am afraid there is not much you can do. To improve the time cost, it is required to profile the stopping process, find the bottleneck and fix it. @FE-EnricoSartori

HumbertoE, 2023-02-17:
Thank you for the verification. Please let me know if there is some improvement in this regard.

Maverobot, 2023-02-28:
@HumbertoE Sorry for the misleading information. After a second check, it turned out that the stopping might not be the (only) source of the delay. I opened a PR which significantly reduces the delay of calling error recovery. Can you please check if it fixes your issue?

HumbertoE, 2023-03-01:
Good to hear. I will then run the experiment again, but with the Franka ROS on commit 2d458abad7390bb73771c91787c361c2b5cfa6ad, after the mentioned PR.

I will report then if the times are improved or not

HumbertoE, 2023-03-01:
I did the experiment again with the [PR](https://github.com/frankaemika/franka_ros/pull/279) of #232 and unfortunately still got similar durations as the ones reported previously. Here there is a table with the new durations.

Timestamp of 1 [s] | Timestamp of 2 [s] | Timestamp of 3 [s] | Duration from 1 to 2 [s] | % | Duration from 2 to 3 [s] | % |  Duration from 1 to 3 [s]
-- | -- | -- | -- | -- | -- | -- | --
1677673485.6275 | 1677673485.62889 | 1677673492.96849 | 0.0014 | 0.02% | 7.3396 | 99.98% | 7.3410
1677673816.95238 | 1677673816.95447 | 1677673819.68224 | 0.0021 | 0.08% | 2.7278 | 99.92% | 2.7299
1677674038.47682 | 1677674038.47934 | 1677674039.17731 | 0.0025 | 0.36% | 0.6980 | 99.64% | 0.7005
1677674147.85044 | 1677674147.85261 | 1677674151.59693 | 0.0022 | 0.06% | 3.7443 | 99.94% | 3.7465
1677674269.48645 | 1677674269.48814 | 1677674272.10449 | 0.0017 | 0.06% | 2.6164 | 99.94% | 2.6180
1677674397.38669 | 1677674397.38861 | 1677674399.88418 | 0.0019 | 0.08% | 2.4956 | 99.92% | 2.4975
1677674508.8368 | 1677674508.83859 | 1677674509.58911 | 0.0018 | 0.24% | 0.7505 | 99.76% | 0.7523
1677674643.74671 | 1677674643.74849 | 1677674644.79248 | 0.0018 | 0.17% | 1.0440 | 99.83% | 1.0458
1677674731.5977 | 1677674731.60386 | 1677674732.48572 | 0.0062 | 0.69% | 0.8819 | 99.31% | 0.8880
1677674823.51982 | 1677674823.52105 | 1677674823.68413 | 0.0012 | 0.75% | 0.1631 | 99.25% | 0.1643
average |   |   | 0.0023 | 0.26% | 2.2850 | 99.74% | 2.2874

In case it is useful, I noticed that the robot first stops, the LEDs stay in green for some time, then they change to red and quickly to green again.

This were the logs for the first attempt of the experiment just as an example of how the logs looked. I took the durations from these timestamps.
```
[ERROR] [1677673485.627496941]: libfranka: Move command aborted: motion aborted by reflex! ["joint_reflex", "cartesian_reflex"]
control_command_success_rate: 1
[ INFO] [1677673485.627570161]: franka_control: main loop
[WARN] [1677673485.628891]: Manipulator error detected. Robot mode: REFLEX
[ INFO] [1677673492.968491856]: Recovered from error
[ INFO] [1677673492.968849281]: franka_control: controller activated
[INFO] [1677673492.971180]: Manipulator error confirmed
```

HumbertoE, 2023-03-01:
Are you able to do the experiment and reproduce the long times with the steps I wrote at the beginning? Let me know if you need more information to do so.

It would be better if you could reproduce the issue and check if it is solved in a similar way as I do to avoid me having to test too many things.

Maverobot, 2023-03-01:
Thanks for the quick test. 

This PR should have fixed the long waiting time caused by the mutex locking, which could take up to 5 seconds. Maybe there are also other causes involved. I will try to find time to reproduce it with your steps again next week. Please be aware that we are now in kurzarbeit and only limited capacity is available.

Maverobot, 2023-03-21:
Hi Humberto,

sorry, I think I sent you the wrong PR in my previous comments. Can you please check out the PR #323 ?

closed 2023-04-21 by Maverobot

HumbertoE, 2023-05-05:
Sorry for taking long to check this solution.

I repeated the experiment with the latest franka ros version (https://github.com/frankaemika/franka_ros/commit/358406952597c3735d43625595650901be84a5d8) and it works much better.

I evaluated both parts of the issue that I wrote at the beginning:

> 1\. There is a significant time difference between the manipulator stopping during a trajectory (we carefully blocked the movement of the robot with our hands) and the cartesian reflex being [...]

### Part 2
For part 2 I evaluated the timestamps of when a print was done for the following moments:

> 1\. Cartesian Reflex, triggered by human interaction, is logged by libfranka with `Move command aborted....`
> 
> 2\. My script's subscriber callback is triggered and logs `Manipulator error detected. Robot mode: REFLEX`. The error recovery action is called immediately after this.
> 
> 3\. `Recovered from error` is logged, I think by the franka_control_node

I tried it with 8 movements and got the following numbers:

Duration from 1 to 2 [s] | % | Duration from 2 to 3 [s] | % |  Duration from 1 to 3 [s]
-- | -- | -- | -- | --
0.0017 | 2.36% | 0.0696 | 97.64% | 0.0713
0.0017 | 1.62% | 0.1022 | 98.38% | 0.1039
0.0015 | 11.21% | 0.0122 | 88.79% | 0.0138
0.0049 | 5.19% | 0.0895 | 94.81% | 0.0944
0.0066 | 44.55% | 0.0082 | 55.45% | 0.0148
0.0021 | 8.87% | 0.0215 | 91.13% | 0.0236
0.0015 | 3.39% | 0.0414 | 96.61% | 0.0429
0.0043 | 23.45% | 0.0139 | 76.55% | 0.0182

with the averages being:

Duration from 1 to 2 [s] | % | Duration from 2 to 3 [s] | % |  Duration from 1 to 3 [s]
-- | -- | -- | -- | --
0.0030 | 12.58% | 0.0448 | 87.42% | 0.0479

So now it is not only much more consistent and it is not taking 3 or 5 seconds, but it's also quite fast. Thank you for the solution.

### Part 1
We did recorded the experiment and reviewing the video, found out that it required around 1.9s between the robot stopped and the error appeared in the terminal. The motors sounded active a bit after the robot stopped and until the error appeared (this also can be heard a bit in the video). I will send you the video by mail @Maverobot.

I don't know if this time is required to stop the motors. Either way for us the biggest problem of these two was part 2. Part 1 does not affect us much so it is not time critical for us, but still wanted to mention it.

### Conclusion
Part 2 is solved and part 1 could be improved, but it does not affect us at the moment, so for us the issue can be closed

PR #323 (2023-04-21): fix: Unexpected long delay when calling error recovery (#317)
Locking of mutex in the busy loop causes long delay when calling error recovery because executing error recovery requires the same mutex. Fixes #317.