frankarobotics/franka_ros #325: Possible race condition in getMass()

nicolapiccinelli, 2023-03-01:
Current setup:

Firmware version: 4.2.2
libfranka version: 0.9.2
franka_ros version: 0.9.1

I'm facing strange behaviour while using the mass matrix to perform inverse dynamic control. It seems that if I don't assign the Eigen map type `Eigen::Map<Eigen::Matrix<double, 7, 7>>` to a Matrix type `Eigen::Matrix<double, 7, 7>` I'm not able to get a stable control behaviour.

Maybe there's a race condition running on the retrieval of the mass matrix values? You can also get a working behaviour if you for instance do `mass.transposeInPlace()`. Obviously, since the matrix should be symmetric that transpose should not change the matrix's values but maybe is doing some implicit conversion.

This seems to not be affecting the **getCoriolis()** and the **getZeroJacobian()** function calls. 

NOT Working:
```
Eigen::Map<Eigen::Matrix<double, 7, 7>> mass(model_handle_->getMass().data());
Eigen::Map<Eigen::Matrix<double, 7, 1>> coriolis(model_handle_->getCoriolis().data());
Eigen::Map<Eigen::Matrix<double, 6, 7>> jacobian(model_handle_->getZeroJacobian(franka::Frame::kEndEffector).data());
```

Working:
```
Eigen::Matrix<double, 7, 7> mass = Eigen::Map<Eigen::Matrix<double, 7, 7>>(model_handle_->getMass().data());
Eigen::Map<Eigen::Matrix<double, 7, 1>> coriolis(model_handle_->getCoriolis().data());
Eigen::Map<Eigen::Matrix<double, 6, 7>> jacobian(model_handle_->getZeroJacobian(franka::Frame::kEndEffector).data());
```

nicolapiccinelli, 2023-03-03:
@Maverobot  the problem seems also to affect the values of the Jacobian matrix. The console output in this case is the following:

```
Jacobian before: 
  -0.471386   0.0535294   -0.543181 -0.00506976  -0.0930387   0.0166489           0
  0.0516769    0.318784   0.0648949  0.00372688    0.011941   0.0907952           0
          0   -0.473435 -0.00779771    0.485942  0.00449933    0.103306           0
          0   -0.986193  -0.0476534    0.991673    0.128085    0.990725  -0.0273559
          0    0.165599   -0.283791   -0.128285     0.99162   -0.127154    0.158733
          1           0    0.957701   0.0113298   0.0168713  -0.0479113   -0.986943

Jacobian after: 
6.95318e-310     -2.49161 2.96439e-323            0            0            0            0
           0      6.77278            0            0            0            0            0
    0.975511  4.6702e-310            1            0            0            0            0
    -0.38814 3.45846e-323            0     -1.04064       6.0383            1   -0.0273559
    -1.34084 6.95318e-310           -0    -0.985472    -0.169836           -0     0.158733
     8.31596            0            0      -1.0544      6.11819            0    -0.986943
```

Maverobot, 2023-03-03:
Hi Nicola,

thanks for writing such a detailed issue. However, I think the issue is not related to franka_ros or libfranka. 

From the **not working** code snippet, 

```cpp
Eigen::Map<Eigen::Matrix<double, 7, 7>> mass(model_handle_->getMass().data());
```

`Eigen::Map` mass stores internally the pointer pointing to the underlying raw array of `std::array<double, 49>` (the mass matrix) returned by `model_handle_->getMass()`, whose lifetime ends after the line finishes. 

Therefore, if you try to access the data by the pointer, it can be / not be the actual mass matrix because the memory, where the mass matrix used to occupy, can be now written with other data.

With your **working** code snippet, the data is copied into a Eigen::Matrix and its lifetimes ends at the scope end.

closed 2023-03-03 by Maverobot

nicolapiccinelli, 2023-03-03:
@Maverobot  what you are saying is correct, but the lifespan of the raw arrays is not documented or at least I was not able to find any information about it. Moreover, in all the examples controller in **franka_ros** the maps are extensively used. So imho if the lifespan of the array is not maintained during the execution of the update function should be mentioned and maybe not used in your examples.

Maverobot, 2023-03-03:
> what you are saying is correct, but the lifespan of the raw arrays is not documented or at least I was not able to find any information about it. 

The lifespan of the raw array was limited to that line because its owner - the std::array - is a temporary value. The lifetime of objects or references is defined as part of the specification of the C++ programming language. For your interest, please check the "Temporary object lifetime" section in https://en.cppreference.com/w/cpp/language/lifetime

> Moreover, in all the examples controller in franka_ros the maps are extensively used. So imho if the lifespan of the array is not maintained during the execution of the update function should be mentioned and maybe not used in your examples.

They are used for demonstration purposes. We assume that if the user is not familiar with `Eigen::Map`, he/she can look it up by themselves. Explaining how `Eigen` works should not be part of the our concerns.

Maverobot, 2023-03-03:
> have temporary variables in the model_handle and class variables from the state_handle

Still these can be check by reading the header files. 

Please explain it more specifically if I am missing something.

nicolapiccinelli, 2023-03-03:
> It was never my intention to teach you C++. Merely trying to help you clearing your doubts.
> 
> > have temporary variables in the model_handle and class variables from the state_handle
> 
> Still these can be check by reading the header files.
> 
> Please explain it more specifically if I am missing something.

I can read the header files or look at your example codes and try to derive some coherence in what is in general for the end-users supposed to be best practices. Also because since I know what an Eigen map does, if you use it extensively in almost your code I think it's reasonable to assume that those data would stay coherent during the update function.

But this is not the case and this behaviour is reasonable for your side, good to know. I hope this thread will help someone else in the future.