frankarobotics/franka_ros #327: Change Gravity Vector

ARoefer, 2023-03-16:
Hello!

We would like to mount our arm at a slightly inclined angle and thus would like to change its gravity vector. In the docs (https://frankaemika.github.io/docs/franka_ros.html#frankastateinterface) it says that `O_ddP_0` will be equal to the `gravity_vector`, however passing this param to `franka_control_node` does not seem to do anything. What are we doing wrong?

laozizhou, 2023-04-17:
> The docs you are referring to are for Gazebo. So you should be able to put the arm at an inclined angle in the simulation without problems. **You cannot change the gravity vector on the real robot!** However, if the angle really is tiny you can try to use the model to subtract the gravity torques in your torque controller for a gravity vector of {0,0,-9.81} (this "removes" the gravity compensation for the robot) and then add the gravity torques that you calculate with the model for your new gravity vector to "add" gravity compensation again. But keep in mind that we neither test this nor support this.

Thanks for the explanation! Can I check with you that where can I change the gravity compensation when using effort_controllers/joint_trajectory_controller?

marcbone, 2023-04-18:
I would try to implement this change in franka_hw. If I would need to do this I would try  to implement the subtraction and addition [here](https://github.com/frankaemika/franka_ros/blob/7a4fcdb6c0d47cea81c1ac7960f30ed8bf85b5fe/franka_hw/src/franka_hw.cpp#L390). You can add a `franka::Model` member variable to the class and then use its gravity method to calculate the needed gravity torques. Be very careful though! I want to say once again that the robot is only intended for table top mounting and that you are risking to damage your robot.

laozizhou, 2023-04-19:
> I would try to implement this change in franka_hw. If I would need to do this I would try to implement the subtraction and addition [...]

The suggestions are of great help. Really appreciate it!