frankarobotics/franka_ros #328: libfranka: Attempt to set invalid transformation in motion generator. Has to be column major

Sophia-study, 2023-03-20:
**Context**
ubuntu: melodic
libfranka: 0.8.0
franka_ros: 0.8.0

**Problem**
Error launching cartesian_pose_example_controller.launch file, when I want to understand cartesian_pose_handle. I don't understand why this example I'm running is wrong, is there anything I haven't thought of?

`[ INFO] [1679307925.856779009]: FrankaHW: Prepared switching controllers to cartesian_pose with parameters limit_rate=1, cutoff_frequency=100, internal_controller=joint_impedance`
`[INFO] [1679307925.862252]: Started controllers: cartesian_pose_example_controller`
`[INFO] [1679307925.866132]: Started controllers: franka_state_controller`
`terminate called after throwing an instance of 'std::invalid_argument'`
`what():  libfranka: Attempt to set invalid transformation in motion generator. Has to be column major!`

marcbone, 2023-03-21:
Thanks for your reply. At least the new pose is homogeneous. I think if  you go to this [config file](https://github.com/frankaemika/franka_ros/blob/0.8.0/franka_control/config/franka_control_node.yaml) and set rate_limiting to false and the cutoff_frequency to 1000 it will work.

Sophia-study, 2023-03-21:
Thanks a lot for your reply, it's working fine now.

closed 2023-03-21 by Sophia-study