frankarobotics/franka_ros #332: Issues while connecting to FCI

sahanant, 2023-04-05:
I am currently trying to work on the Franka Emika Research 3 Arm.  While I can control and move the arm wit with the help of its desk interface, currently I am facing an issue with regards to connecting with it through ros. 
While trying to activate FCI on the desk interface, I am prompted with an error **"MonitoredStandStill"**. It is my first time working with this arm and it would be extremely helpful if any of you could help me solve the issues. If this is not the right place to post this issue I apologize in advance.

For reference, I have attached the error image down below.

[image]

khaninger, 2023-04-06:
I have the same issue. Current workaround is triggering the external stop, then releasing it, then activating FCI. However, when I do this, the echo_robot_state  returns "User Stopped", so I'm not sure it's a good workaround.

sahanant, 2023-04-06:
> I have the same issue. Current workaround is triggering the external stop, then releasing it, then activating FCI. However, when I do this, the echo_robot_state returns "User Stopped", so I'm not sure it's a good workaround.

I followed the steps and was able to activate FCI. The only thing I would like to add is that I had to wait till the gripper started its initialization process to activate the FCI. That is, I was only able to activate the FCI when the gripper was in the process of initializing . Before and after the gripper initialization process, Activating FCI yields the same error.

andrejpan, 2023-04-11:
There is also additional information pasted on [FE forume](https://www.franka-community.de/t/libfranka-move-command-rejected-command-not-possible-in-the-current-mode-user-stopped/4454).

marcbone, 2023-04-11:
I investigated a bit and the root problem is that you have the safety functions activated. However, you should still be able to activate FCI and only get an error when you try to move the robot with FCI while a safety function is activated. That you cannot even enable FCI in Desk is a newly introduced bug in the user interface. I notified the responsible team. In the meantime you can either delete your safety rules or use your workaround by activating FCI during reinitialization of the gripper

sahanant, 2023-04-13:
> Your error is probably because you have safety functions activated. If you are only interested in using FCI, you can disable them using the safety operator account in watchman. It should be documented in the manual. Also take a look [here](https://www.franka-community.de/t/libfranka-move-command-rejected-command-not-possible-in-the-current-mode-user-stopped/4454/10). You should be able to activate FCI at any time. Therefore, I am reopening this. Issues like this should usually not be opened here and rather go directly to [support@franka.de](mailto:support@franka.de) since it is not a problem with franka_ros. On the other side this might also help others

I have implemented the given steps in the forum and was successful in activating FCI and establishing control of the arm.  For any further queries regarding the robot arm I will refer to the franka emika community page. Thanks @andrejpan @marcbone  for guiding me to the solution

closed 2023-04-13 by sahanant