frankarobotics/franka_ros #336: Dual arm combined control for FR3

BolunDai0216, 2023-04-14:
Hi,

I was looking at how to control two FR3s using `franka_ros`, and it seems like now it is not well supported. Can someone help me understand what work needs to be done to make this happen?

From my understanding, it would require 
- adding a URDF file for the two FR3 settings in `franka_description`;
- change the config files `franka_combined_control_node.yaml` and `default_combined_controllers.yaml` in `franka_control`;
- change the launch file `franka_combined_control.launch` in `franka_control` to allow switching between fr3 and panda.

On top of these two changes, if I want to make an FR3 version of the `dual_arm_cartesian_impedance_example_controller`, I think I would need to change
- `franka_example_controllers.yaml`;
- `dual_arm_cartesian_impedance_example_controller.launch`.

If I also want to change the `teleop_joint_pd_example_controller`, I think the files that need to be updated are
- `franka_example_controllers.yaml`
- `teleop_joint_pd_example_controller.launch`
- `teleop_joint_pd_example_controller.cpp`

Is that correct? I also find it hard to believe that this is not already there. Is it planned for the next release?

Thanks in advance!

andrejpan, 2023-04-16:
Hey @BolunDai0216,
dual arm control with FR3 was not carried off so far at FE and there is no plan to support it in near future. 

Hopefully, the community can provide some support on this issue.

BolunDai0216, 2023-04-17:
@andrejpan Are there currently any hardware or software limitations to achieving dual arm control with FR3?

andrejpan, 2023-04-17:
Afaik, no. It is a niche setup, and we did have time to replicate it.

marcbone, 2023-04-17:
Dual arm control should work with FR3. You can use the current examples for panda that we have. They also work with FR3. However, for better results you might want to replace the "joint_limits.yaml" file from panda with the "joint_limits.yaml" file from FR3 in the [dual_panda_example.urdf.xacro file](https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/dual_panda/dual_panda_example.urdf.xacro)

BolunDai0216, 2023-04-17:
@marcbone the panda examples also works for my dual FR3 settings, thanks for pointing that out!