frankarobotics/franka_ros #338: How to realize the gravity compensation when the robot is tilted mounting?

HongminWu, 2023-04-22:
When we install the robot in this tilted way, there are two problems as follows: 
(1) when the white light is displayed, it is particularly difficult to drag and not smooth; 
(2) we want to control the robot by ROS commands, but can't keep the robot in a state of blue light. 
We guess it is the problem of gravity compensation caused by the tilted installation. 
Please help us.
[image]

andrejpan, 2023-04-24:
Hi @HongminWu 
1. In the manual is stated that the robot works appropriately if it is mounted horizontally up. For example, the guiding mode needs to compensate for gravity force. When you tilted the robot like this, gravity compensation does not work correctly, ergo you are observing described issues. There is [support](https://www.mirmi.tum.de/mirmi/research/work/kifabrik-bayern/) for tilted FER, however I am not sure if it is publicly available. Please send an email about this query to support@franka.de

2. Similarly goes for running FCI, you need to offset different gravity vector based on how you tilted the robot. There were quite a few similar topics here and [FE forum](https://www.franka-community.de/). Please check them for more information.

marcbone, 2023-04-24:
Also related: #327