frankarobotics/franka_ros #340: Why no gravity term when computing tau in cartesian_impedance_example_controller?

Yitian-Xian, 2023-05-08:
Hello, may I know why there is no gravity term when computing 'tau_d'?

This term is added by tau_d << tau_task + tau_nullspace + coriolis, which are from Cartesian PD control, nullspace PD control, and the coriolis effect.

Could anyone explain that? Thank you very much.

rickstaa, 2023-09-01:
Hey @Yitian-Xian,

I just noticed that your last response got lost in my mailbox. I hope you've managed to resolve your issues, and I apologize for the delay in getting back to you 😅. If it's okay with you, I'll provide an answer here so that others can refer to it as well. However, please keep in mind that I'm not part of the @frankaemika team; I'm just an end user. So, there's a chance I may have misunderstood certain aspects.

Regarding your question in the email:

Does this mean that gravity is always compensated in simulation, so we don't need to include the gravity term when calculating the torque? I've reviewed all the examples in the franka_example_controllers package, and I haven't found any of them including the gravity term.

You are correct in observing that in the simulation (specifically, franka_gazebo), gravity compensation is applied. You can see it in the code snippet here:

https://github.com/frankaemika/franka_ros/blob/d439fc7d66a3edb890eefb0fa955f6d7042cac12/franka_gazebo/src/franka_hw_sim.cpp#L512

As a result, if you also account for gravity when calculating the desired torque (e.g., tau_d), as seen in the `cartesian_impedance_example_controller` code snippet here:

https://github.com/frankaemika/franka_ros/blob/553e18217ef3c7b702872c7f67450f832a3d7952/franka_example_controllers/src/cartesian_impedance_example_controller.cpp#L181

You would essentially be adding the gravity component twice. From what I recall in my past conversation with @gollth, he explained that for the real robot, gravity compensation is handled within the franka control unit, leading to the same conclusion.

An important point to note is that, for the simulation, when `tau_external` is calculated, you do need to include the gravity component:

https://github.com/frankaemika/franka_ros/blob/553e18217ef3c7b702872c7f67450f832a3d7952/franka_gazebo/src/franka_hw_sim.cpp#L493

This is because the code snippet mentioned above applies the gravity component just before the control is applied to the simulation. Consequently, the value returned by the self->effort member doesn't include the gravity.

If there are any inaccuracies in my response, @gollth, @marcbone, or @Maverobot feel free to correct me 👍🏻.

gollth, 2023-09-04:
Thanks @rickstaa, you nailed it perfectly =)