frankarobotics/franka_ros #341: Safety Violation Error Triggered by External Abrupt Movements

HumbertoE, 2023-05-10:
When an external abrupt movement is applied to the robot, a safety violation error occurs, prompting the message "Please confirm the safety violation(s) have been resolved." as shown in the following screenshot:

[image]

And this is the rule 2 description under "work" safety scenario as stated in the generated Safetyexport:
[image]

This error first occurred when the mobile platform to which the robot is mounted passed above something, causing an abrupt movement. Upon further testing, we observed that the error consistently happens when we move one of the robot's axes quickly but only a little bit. Interestingly, if we move it slowly, even with the same small distance, the error doesn't occur.  The error happens regardless of whether the robot is locked or unlocked and whether the Force Control Interface (FCI) is enabled or disabled.

The main problem we have is that we cannot confirm this error programmatically through ROS.

We need help identifying the cause of this error and finding a solution. Since the mobile platform won't always move on smooth surfaces, we expect to come across this issue quite often. Any insights, suggestions, or fixes to eliminate this error and ensure safe robot operations would be greatly appreciated.

andrejpan, 2023-05-10:
Can you show the complete rule? We need info about activation space and safety function. 
[image].
Currently, you could easily set up a rule that will always engage safety violations if the brakes are locked or not.

FCI currently (and probably in the future) will not be able to modify or resolve safety-relevant issues. Resolving these issues can only be done via Desk.

andrejpan, 2023-05-10:
These two rules are almost default ones. You should get in touch with our customer support since this looks like an internal problem with a robot.

HumbertoE, 2023-08-28:
Abrupt movements cause accelerations on the robot that trigger the SMS. This makes sense and is the desired behavior so I will close the issue.
Thank you.

closed 2023-08-28 by HumbertoE